pioarduino/platform-espressif32 #10: fatal error: esp_eth_driver.h: No such file or directory

plystdr, 2024-07-21:
Throws an error during build.

```
In file included from C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:29,
                 from C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:33,
                 from src/main.cpp:14:
C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:35:10: fatal error: esp_eth_driver.h: No such file or directory

************************************************************************
* Looking for esp_eth_driver.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_eth_driver.h"
* Web  > https://registry.platformio.org/search?q=header:esp_eth_driver.h
*
************************************************************************

   35 | #include "esp_eth_driver.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\lgfx_fonts.cpp.o
*** [.pio\build\xxx\src\main.cpp.o] Error 1
```

Here is the platformio.ini below
```
[common]
lib_deps = 
	https://github.com/joltwallet/esp_littlefs.git
	ayushsharma82/ElegantOTA@^3.1.3

[env:xxx]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip
framework = espidf, arduino
board = xxx
board_build.mcu = esp32s3
upload_protocol = esptool
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.partitions = partition_scheme.csv
board_build.filesystem = littlefs
board_upload.offset_address = xxx
lib_deps = 
	${common.lib_deps}
extra_scripts = 
	tasks.py
build_flags = 
	-DCORE_DEBUG_LEVEL=5
	-DBOARD_HAS_PSRAM
	-DARDUINO_ESP32S3_DEV
	-DCONFIG_SPIRAM_USE
	-mfix-esp32-psram-cache-issue
	-w
	-Wno-unused-variable
	-Wno-unused-function
	-DELEGANTOTA_USE_ASYNC_WEBSERVER=1
```

mathieucarbou, 2024-07-21:
I think this is a change related to Arduino 3.0.3, not pioarduino. 
I had also myself to  include additional headers when switching to Arduino 3.0.3

I see you are using `ayushsharma82/ElegantOTA@^3.1.3`: make sure to use:

```
  mathieucarbou/AsyncTCP @ 3.2.3
  mathieucarbou/ESPAsyncWebServer @ 3.1.1
  ayushsharma82/ElegantOTA @ 3.1.3
```

It works for me.

Jason2866, 2024-07-21:
Remove this lib. It is already there :-) Adding can do harm.
`https://github.com/joltwallet/esp_littlefs.git`

Jason2866, 2024-07-21:
Closing issue since it is NOT an framework problem.

closed 2024-07-21 by Jason2866

plystdr, 2024-07-21:
No change. PlatformIO can grab those 2 libraries even without putting them there.
Not sure what is going on. If I use older platformio libraries, it compiles fine.

Here is the full output with updated platformio.ini.
i cleaned everything in packages folder and downloaded everything from scratch

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[common]
lib_deps = 
	mathieucarbou/AsyncTCP @ 3.2.3
 	mathieucarbou/ESPAsyncWebServer @ 3.1.1
	ayushsharma82/ElegantOTA@^3.1.3

[env:polystruder-grpro-r31-recover]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip
framework = espidf, arduino
board = polystruder-grcircuit
board_build.mcu = esp32s3
upload_protocol = esptool
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.partitions = partition_scheme.csv
board_build.filesystem = littlefs
board_upload.offset_address = 0x670000
lib_deps = 
	${common.lib_deps}
extra_scripts = 
	tasks.py
build_flags = 
	-DCORE_DEBUG_LEVEL=5
	-DBOARD_HAS_PSRAM
	-DARDUINO_ESP32S3_DEV
	-DCONFIG_SPIRAM_USE
	-w
	-Wno-unused-variable
	-Wno-unused-function
	-DELEGANTOTA_USE_ASYNC_WEBSERVER=1

```

```
Executing task: C:\Users\xxx\.platformio\penv\Scripts\platformio.exe run --environment xxx 

Processing xxx (platform: https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip; framework: espidf, arduino; board: xxx)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/xxx.html
PLATFORM: Espressif 32 (51.3.3) > Espressif ESP32-S3-N8-R2 (8 MB QD, 2MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.3 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.dc859c1e67 
 - framework-espidf @ 3.50104.0 (5.1.4) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.7.5 
 - tool-idf @ 1.0.1 
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
[... 46 lines cut ...]
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\panel\Panel_SSD1327.cpp.o
In file included from C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:29,
                 from C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:33,
                 from src/main.cpp:14:
C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:35:10: fatal error: esp_eth_driver.h: No such file or directory

************************************************************************
* Looking for esp_eth_driver.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_eth_driver.h"
* Web  > https://registry.platformio.org/search?q=header:esp_eth_driver.h
*
************************************************************************

   35 | #include "esp_eth_driver.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\panel\Panel_SSD1331.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\panel\Panel_SSD1351.cpp.o
*** [.pio\build\xxx\src\main.cpp.o] Error 1
======================================================================== [FAILED] Took 12.90 seconds ========================================================================

 *  The terminal process "C:\Users\xxx\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'xxx'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

plystdr, 2024-07-21:
found the problem.
adding esp-idf framework is causing the issue itself.
however it does not solve the issue, since i need both arduino and espidf frameworks at the same time.

`framework = espidf, arduino`

this time i got different errors, but these are due to no espidf framework is present.
```
Executing task: C:\Users\xxx\.platformio\penv\Scripts\platformio.exe run --environment xxx 

Processing xxx (platform: https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip; framework: arduino; board: xxx)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/xxx.html
PLATFORM: Espressif 32 (51.3.3) > Espressif ESP32-S3-N8-R2 (8 MB QD, 2MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.3
 - framework-arduinoespressif32-libs @ 5.1.0+sha.dc859c1e67
 - tool-esptoolpy @ 4.7.5
 - tool-mklittlefs @ 3.2.0
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823 
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP @ 3.2.3
|-- ESPAsyncWebServer @ 3.1.1
|-- ElegantOTA @ 3.1.3
|-- HTTPClient @ 3.0.3
|-- Update @ 3.0.3
|-- WiFi @ 3.0.3
|-- ui
|-- LovyanGFX @ 1.1.12
|-- lvgl @ 8.3.11
Building in release mode
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\Fonts\IPA\lgfx_font_japan.c.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\Fonts\efont\lgfx_efont_cn.c.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\Fonts\efont\lgfx_efont_ja.c.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\Fonts\efont\lgfx_efont_kr.c.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\Fonts\efont\lgfx_efont_tw.c.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\utility\lgfx_pngle.c.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\utility\lgfx_qoi.c.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\utility\lgfx_qrcode.c.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\utility\lgfx_tjpgd.c.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\utility\miniz.c.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v0\LGFXBase.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v0\lgfx_fonts.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v0\panel\Panel_GDEW0154M09.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v0\panel\Panel_IT8951.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v0\platforms\esp32_common.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v0\platforms\samd51_common.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v0\touch\Touch_FT5x06.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v0\touch\Touch_GT911.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v0\touch\Touch_STMPE610.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v0\touch\Touch_XPT2046.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\LGFXBase.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\LGFX_Button.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\LGFX_Sprite.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\lgfx_fonts.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\misc\DividedFrameBuffer.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\misc\SpriteBuffer.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\misc\common_function.cpp.o
Compiling .pio\build\xxx\libbdc\LovyanGFX-1.1.12\lgfx\v1\misc\pixelcopy.cpp.o
[... 134 lines cut ...]
      |          IP_EVENT_STA_GOT_IP
include/wifiConnect.h:91:10: error: 'SYSTEM_EVENT_STA_LOST_IP' was not declared in this scope; did you mean 'IP_EVENT_STA_LOST_IP'?
   91 |     case SYSTEM_EVENT_STA_LOST_IP:
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
      |          IP_EVENT_STA_LOST_IP
Compiling .pio\build\xxx\libc75\Update\Updater.cpp.o
include/wifiConnect.h:97:10: error: 'SYSTEM_EVENT_STA_DISCONNECTED' was not declared in this scope; did you mean 'WIFI_EVENT_STA_DISCONNECTED'?
   97 |     case SYSTEM_EVENT_STA_DISCONNECTED:
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |          WIFI_EVENT_STA_DISCONNECTED
include/wifiConnect.h:107:10: error: 'SYSTEM_EVENT_STA_CONNECTED' was not declared in this scope; did you mean 'WIFI_EVENT_STA_CONNECTED'?
  107 |     case SYSTEM_EVENT_STA_CONNECTED:
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |          WIFI_EVENT_STA_CONNECTED
Compiling .pio\build\xxx\libf83\ElegantOTA\ElegantOTA.cpp.o
Archiving .pio\build\xxx\libf2b\libFS.a
Indexing .pio\build\xxx\libf2b\libFS.a
Compiling .pio\build\xxx\libf83\ElegantOTA\elop.cpp.o
*** [.pio\build\xxx\src\main.cpp.o] Error 1
======================================================================== [FAILED] Took 24.29 seconds ========================================================================

 *  The terminal process "C:\Users\xxx\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'xxx'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

Jason2866, 2024-07-21:
It is a code issue. The CI does compile successfully all examples. Arduino, espidf and Arduino as an component of IDF. May you look at the example folders, in this repo.

italocjs, 2024-12-01:
Hello, how did you fix it? i've been unable to use both arduino and espidf as frameworks too. 

my platformio.ini > https://pastebin.com/J98FpDNp
the error logs > https://pastebin.com/R7Ts4eJE

I have been trying to move forward for hours without any luck.

Jason2866, 2024-12-01:
@italocjs This should work. Don't change the compiler version! This is not supported from IDF nor Arduino
 ```
[env:esp32dev]
; build_type = debug
; debug_build_flags = -Os
;lib_ldf_mode = deep+
;test_framework = googletest
;test_ignore = test_native
platform = https://github.com/pioarduino/platform-espressif32.git#develop
board = simovatrack130
framework = arduino, espidf
 
board_build.partitions = partitions.csv
;build_unflags = -std=gnu++11
build_flags = 
    -D USE_ESP32_ENVIRONMENT
;    -std=gnu++17
    -Wall                ; Enable all common warnings
    -Wunused-function    ; Warn about unused functions
board_build.embed_txtfiles =
    managed_components/espressif__esp_insights/server_certs/https_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt
 
upload_speed = 921600
monitor_speed = 115200
; monitor_filters = esp32_exception_decoder 
monitor_filters = 
    esp32_exception_decoder
    send_on_enter
 
lib_deps = 
    adafruit/Adafruit Unified Sensor @ 1.1.14
    adafruit/Adafruit BusIO @ 1.16.1
    adafruit/Adafruit ADXL343 @ 1.6.4
    https://github.com/italocjs/DFRobot_LIS.git @ 1.0.1+sha.5d34332
    https://github.com/italocjs/geofencing_esp32.git @ 1.0.0+sha.140dd27
    https://github.com/italocjs/NuS-NimBLE-Serial @ 2.0.4+sha.9ea1cdb
    mprograms/QMC5883LCompass @ 1.2.3
    adafruit/RTClib @ 2.1.4
    mikalhart/TinyGPSPlus @ 1.1.0
    bblanchon/ArduinoJson @ 7.1.0
    adafruit/Adafruit GFX Library @ 1.11.10
    adafruit/Adafruit SSD1306 @ 2.5.11
    NimBLE-Arduino @ 1.4.2
 
; extra_scripts = esp32_exception_decoder.py
```
Errors which occurs now are caused from libs not compatible!
NimBLE-Arduino does work not for newer MCUs

italocjs, 2024-12-01:
hello @Jason2866 , thanks!  
I removed everything from my code, including all libs, keeping only a minmal printf but still have this issue.  any further ideas? i also cleaned `.plaftormio/packages` , `.plaftormio/platforms` and `.pio` to make sure no cache was causing issues.

I have prepared an detailed pastebin, hope you may help me catch something.  https://pastebin.com/YvkbCRSZ
I need both espidf and arduino frameworks to be able to tweak menuconfig

Jason2866, 2024-12-01:
@italocjs  Probably you are hit from a bug in pio core LDF 
See here https://github.com/platformio/platformio-core/issues/4818