pioarduino/platform-espressif32 #11: "group ended before it began" error during final build step.

plystdr, 2024-07-22:
I managed to fix remaining compilation errors of migrating our project from arduino core 2.0 to 3.0.

however, we are not quite figure out how to resolve this issue.

platformio.ini
[common]
lib_deps = 
	mathieucarbou/AsyncTCP @ 3.2.3
 	mathieucarbou/ESPAsyncWebServer @ 3.1.1
	ayushsharma82/ElegantOTA@^3.1.3
board_build.embed_txtfiles =
    managed_components/espressif__esp_insights/server_certs/https_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
 managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt

[env:xxx]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip
framework = espidf, arduino
board = xxx
board_build.mcu = esp32s3
upload_protocol = esptool
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.partitions = partition_scheme.csv
board_build.filesystem = littlefs
board_upload.offset_address = 0x670000
board_build.embed_txtfiles =
	${common.board_build.embed_txtfiles}
lib_deps = 
	${common.lib_deps}
extra_scripts = 
	tasks.py
build_flags = 
	;-DCORE_DEBUG_LEVEL=5
	-DBOARD_HAS_PSRAM
	-DARDUINO_ESP32S3_DEV
	-DCONFIG_SPIRAM_USE
	-w
	-Wno-unused-variable
	-Wno-unused-function
	-DELEGANTOTA_USE_ASYNC_WEBSERVER=1

```
Building in release mode
Linking .pio\build\xxx\firmware.elf
c:/users/xxx/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: group ended before it began (--help for usage)
```

there is the temp file used in final step
```
-o .pio/build/xxx/firmware.elf -T memory.ld -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T sections.ld -u esp_app_desc -u pthread_include_pthread_impl -u pthread_include_pthread_cond_var_impl -u pthread_include_pthread_local_storage_impl -u pthread_include_pthread_rwlock_impl -u pthread_include_pthread_semaphore_impl -u ld_include_highint_hdl -u start_app -u start_app_other_cores -u __ubsan_include -u __assert_func -u app_main -u newlib_include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u newlib_include_assert_impl -u __cxa_guard_dummy -u __cxx_fatal_exception -u include_esp_phy_override -u vfs_include_syscalls_impl -Wl,--Map=D:/firmware/xxx/.pio/build/xxx/xxx.map -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--end-group -Wl,--gc-sections -Wl,--no-warn-rwx-segments -Wl,--start-group -Wl,--undefined=FreeRTOS_openocd_params -Wl,--undefined=uxTopUsedPriority -Wl,--warn-common -Wl,--wrap=_Unwind_Backtrace -Wl,--wrap=_Unwind_DeleteException -Wl,--wrap=_Unwind_FindEnclosingFunction -Wl,--wrap=_Unwind_Find_FDE -Wl,--wrap=_Unwind_ForcedUnwind -Wl,--wrap=_Unwind_GetCFA -Wl,--wrap=_Unwind_GetDataRelBase -Wl,--wrap=_Unwind_GetGR -Wl,--wrap=_Unwind_GetIP -Wl,--wrap=_Unwind_GetIPInfo -Wl,--wrap=_Unwind_GetLanguageSpecificData -Wl,--wrap=_Unwind_GetRegionStart -Wl,--wrap=_Unwind_GetTextRelBase -Wl,--wrap=_Unwind_RaiseException -Wl,--wrap=_Unwind_Resume -Wl,--wrap=_Unwind_Resume_or_Rethrow -Wl,--wrap=_Unwind_SetEnableExceptionFdeSorting -Wl,--wrap=_Unwind_SetGR -Wl,--wrap=_Unwind_SetIP -Wl,--wrap=__cxa_call_unexpected -Wl,--wrap=__deregister_frame_info -Wl,--wrap=__deregister_frame_info_bases -Wl,--wrap=__gxx_personality_v0 -Wl,--wrap=__register_frame -Wl,--wrap=__register_frame_info -Wl,--wrap=__register_frame_info_bases -Wl,--wrap=__register_frame_info_table -Wl,--wrap=__register_frame_info_table_bases -Wl,--wrap=__register_frame_table -Wl,--wrap=esp_log_write -Wl,--wrap=esp_log_writev -Wl,--wrap=longjmp -fno-lto -fno-rtti -mlongcalls .pio/build/xxx/src/main.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/Fonts/IPA/lgfx_font_japan.c.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/Fonts/efont/lgfx_efont_cn.c.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/Fonts/efont/lgfx_efont_ja.c.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/Fonts/efont/lgfx_efont_kr.c.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/Fonts/efont/lgfx_efont_tw.c.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/utility/lgfx_pngle.c.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/utility/lgfx_qoi.c.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/utility/lgfx_qrcode.c.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/utility/lgfx_tjpgd.c.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/utility/miniz.c.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v0/LGFXBase.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v0/lgfx_fonts.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v0/panel/Panel_GDEW0154M09.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v0/panel/Panel_IT8951.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v0/platforms/esp32_common.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v0/platforms/samd51_common.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v0/touch/Touch_FT5x06.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v0/touch/Touch_GT911.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v0/touch/Touch_STMPE610.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v0/touch/Touch_XPT2046.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/LGFXBase.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/LGFX_Button.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/LGFX_Sprite.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/lgfx_fonts.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/misc/DividedFrameBuffer.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/misc/SpriteBuffer.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/misc/common_function.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/misc/pixelcopy.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_Device.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_FlexibleFrameBuffer.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_FrameBufferBase.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_GDEW0154M09.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_HUB75.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_HasBuffer.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_ILI9225.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_IT8951.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_LCD.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_M5HDMI.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_M5UnitGLASS.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_M5UnitLCD.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_NT35510.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_RA8875.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_RM68120.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_SSD1306.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_SSD1327.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_SSD1331.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_SSD1351.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/panel/Panel_SSD1963.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/arduino_default/Bus_SPI.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/arduino_default/Bus_Stream.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/arduino_default/common.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32/Bus_HUB75.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32/Bus_I2C.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32/Bus_Parallel8.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32/Bus_SPI.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32/Light_PWM.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32/Panel_CVBS.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32/common.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32c3/Bus_Parallel8.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32s2/Bus_Parallel16.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32s2/Bus_Parallel8.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/lgfx/v1/platforms/esp32s3/Bus_Parallel16.cpp.o .pio/build/xxx/libbdc/LovyanGFX-1.1.12/ [... cut ...] brary/libmbedcrypto.a .pio/build/xxx/esp-idf/mbedtls/mbedtls/library/libmbedtls.a .pio/build/xxx/esp-idf/mbedtls/mbedtls/library/libmbedx509.a .pio/build/xxx/esp-idf/mbedtls/mbedtls/3rdparty/p256-m/libp256m.a -lhufzip -lesp_audio_front_end -lesp_audio_processor -lmultinet -lflite_g2p -lnsnet -lwakenet -ldl_lib -lfst -lc_speech_features -lesp_tts_chinese -lvoice_set_xiaole -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lxt_hal -lc -lm -lstdc++ -lgcc -lphy -lbtbb -Wl,--end-group

```

TD-er, 2024-07-22:
Looks like either a [Windows path length](https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry) or command line length error.

You could try to move your `.platformio` folder to a shorter path like this:
```
[platformio]
core_dir = /.platformio
```

Just make sure to also check this folder is excluded from what your virusscanner may scan or else you will run into other really odd build errors.

We should look into the path length issue as I got the feeling paths like these might be possible to shorten a bit :)
```
c:/users/xxx/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin
```

plystdr, 2024-07-22:
Allright so i completely stopped antivirus (kaspersky) temporarily and shorten the foldername of project.

platformio is installed in there. i believe it is short enough.
`C:\Users\alisu\.platformio\`

still getting the same error. here is the verbose build output with unmasked foldernames

```
Executing task in folder grp31_rescue: C:\Users\alisu\.platformio\penv\Scripts\platformio.exe run --verbose --environment grp31_rescue 

Processing grp31_rescue (platform: https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip; framework: arduino, espidf; board: xxx; board_build.mcu: esp32s3; upload_protocol: esptool; monitor_speed: 115200; board_build.f_cpu: 240000000L; board_build.f_flash: 80000000L; board_build.flash_mode: qio; board_build.partitions: partition_scheme.csv; board_build.filesystem: littlefs; board_upload.offset_address: 0x670000; board_build.embed_txtfiles: 
managed_components/espressif__esp_insights/server_certs/https_server.crt
managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt; lib_deps: ayushsharma82/ElegantOTA@^3.1.3, lovyan03/LovyanGFX@^1.1.16; extra_scripts: tasks.py; build_flags: -DCORE_DEBUG_LEVEL=0, -w, -Wno-unused-variable, -Wno-unused-function, -DELEGANTOTA_USE_ASYNC_WEBSERVER=1)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/xxx.html
PLATFORM: Espressif 32 (51.3.3) (https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip) > Espressif ESP32-S3-N8-R2 (8 MB QD, 2MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.3 (https://github.com/espressif/arduino-esp32/releases/download/3.0.3/esp32-3.0.3.zip) 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.dc859c1e67 (https://github.com/espressif/arduino-esp32/releases/download/3.0.3/esp32-arduino-libs-3.0.3.zip) 
 - framework-espidf @ 3.50104.0 (5.1.4, https://github.com/pioarduino/esp-idf/releases/download/v5.1.4.240712/esp-idf-v5.1.4.zip) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.7.5 (https://github.com/pioarduino/esptool/releases/download/v4.7.5/esptool.zip)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mklittlefs @ 3.2.0
 - tool-ninja @ 1.9.0
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
Warning: Ignored `ESPAsyncTCP-esphome` dependency for `ESPAsyncWebServer` library
Warning: Ignored `Hash` dependency for `ESPAsyncWebServer` library
Warning: Ignored `AsyncTCP_RP2040W` dependency for `ESPAsyncWebServer` library
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Warning: Ignored `ESPAsyncTCP-esphome` dependency for `ESPAsyncWebServer` library
Warning: Ignored `Hash` dependency for `ESPAsyncWebServer` library
Warning: Ignored `AsyncTCP_RP2040W` dependency for `ESPAsyncWebServer` library
Dependency Graph
|-- ElegantOTA @ 3.1.3 (License: Unknown, Path: D:\firmware\grp31_rescue\.pio\libdeps\grp31_rescue\ElegantOTA)
|   |-- ESPAsyncWebServer @ 3.1.1 (License: LGPL-3.0, Path: D:\firmware\grp31_rescue\.pio\libdeps\grp31_rescue\ESPAsyncWebServer)
|   |   |-- AsyncTCP @ 3.2.3 (License: LGPL-3.0, Path: D:\firmware\grp31_rescue\.pio\libdeps\grp31_rescue\AsyncTCP)
|   |   |   |-- Networking @ 3.0.3 (License: Unknown, Path: C:\Users\alisu\.platformio\packages\framework-arduinoespressif32\libraries\Network)
|   |   |-- FS @ 3.0.3 (License: Unknown, Path: C:\Users\alisu\.platformio\packages\framework-arduinoespressif32\libraries\FS)
|   |   |-- WebServer @ 3.0.3 (License: Unknown, Path: C:\Users\alisu\.platformio\packages\framework-arduinoespressif32\libraries\WebServer)
|   |   |   |-- FS @ 3.0.3 (License: Unknown, Path: C:\Users\alisu\.platformio\packages\framework-arduinoespressif32\libraries\FS)
|   |   |   |-- Networking @ 3.0.3 (License: Unknown, Path: C:\Users\alisu\.platformio\packages\framework-arduinoespressif32\libraries\Network)
|   |   |-- WiFi @ 3.0.3 (License: Unknown, Path: C:\Users\alisu\.platformio\packages\framework-arduinoespressif32\libraries\WiFi)
|   |   |   |-- Networking @ 3.0.3 (License: Unknown, Path: C:\Users\alisu\.platformio\packages\framework-arduinoespressif32\libraries\Network)
|   |-- AsyncTCP @ 3.2.3 (License: LGPL-3.0, Path: D:\firmware\grp31_rescue\.pio\libdeps\grp31_rescue\AsyncTCP)
|   |   |-- Networking @ 3.0.3 (License: Unknown, Path: C:\Users\alisu\.platformio\packages\framework-arduinoespressif32\libraries\Network)
[... cut ...]
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\grp31_rescue\firmware.elf] Error 1
======================================================================== [FAILED] Took 12.20 seconds ========================================================================

 *  The terminal process "C:\Users\alisu\.platformio\penv\Scripts\platformio.exe 'run', '--verbose', '--environment', 'grp31_rescue'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

plystdr, 2024-07-22:
Additional note that we have other projects that is running on longer path names on standard platformio installation with arduino core 2.0. they compile fine without any errors.

when i check the temp output, i believe it is supposed to put only one  **--start-group** and **--end-group** commands.

however the first command is --end-group followed by --start-group and then another --start-group then --end-group.

Jason2866, 2024-07-22:
The next pioarduino platform release will have shorter folder names for the tools and toolchains. Hopefully this is enough to workaround the Windows path length problem.

Jason2866, 2024-07-23:
Issue is NOT a Windows path length problem. Issue is reproducible.
Compiling example https://github.com/pioarduino/platform-espressif32/blob/develop/examples/espidf-arduino-blink with the S3 generates the error.

Jason2866, 2024-07-23:
@plystdr reopened the issue. It is NOT related to Windows.
You closed the issue, have you found a solution?

plystdr, 2024-07-23:
No. We went back to original platformio core and compiled our firmware in that way.
I will be also waiting for a solution so I can switch to arduino core 3.

ricardoquesada, 2024-07-24:
triggering this issue from Linux + esp-idf 5.1.4 + Arduino Core 3.0.3 as an idf component + esp32-s3

Jason2866, 2024-07-26:
Fixed with commit https://github.com/pioarduino/platform-espressif32/commit/b2404e8b62f1393007a8b940326b2d514e6840ad

Jason2866, 2024-07-26:
@plystdr Please test

ricardoquesada, 2024-07-27:
@Jason2866 how do you test it ?

I've just `rm -rf .pio` and tested it again, and failed with the same error.

My platformio.ini is this one:
https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template/blob/develop/platformio.ini

```
[platformio]
src_dir = main

[env]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip
#platform = espressif32
framework = arduino, espidf
monitor_speed = 115200
monitor_filters = direct
board_build.embed_txtfiles =
    managed_components/espressif__esp_insights/server_certs/https_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt

[env:esp32]
board = esp32dev

# It is not compiling to a linker error
[env:esp32-s3-devkitc-1]
board = esp32-s3-devkitc-1

[env:esp32-c3-devkitc-02]
board = esp32-c3-devkitc-02
```

Jason2866, 2024-07-27:
@ricardoquesada you have to delete all folders and files generated by Platformio in project folder. Restart VSC. The example is tested and verified working. If still a compile error occurs you have overlooked to delete old left reminents.

Jason2866, 2024-07-27:
@ricardoquesada  Just tried with your project https://github.com/ricardoquesada/esp-idf-arduino-bluepad32-template
After fixing the wrong Platformio config (Do **NOT** place Arduino libs in components folder, use lib folder!!)
AND deleting the totally obsolete fork of Arduino in components folder,
it compiles fine for the S3, just a few warnings from your code

```
Linking .pio/build/esp32-s3-devkitc-1/bootloader.elf
Compiling .pio/build/esp32-s3-devkitc-1/lib386/bluepad32_arduino/arduino_bootstrap.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/lib386/bluepad32_arduino/arduino_platform.c.o
Building .pio/build/esp32-s3-devkitc-1/bootloader.bin
esptool.py v4.7.5
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
lib/bluepad32_arduino/arduino_platform.c: In function 'version':
lib/bluepad32_arduino/arduino_platform.c:407:5: warning: 'esp_ota_get_app_description' is deprecated: Please use esp_app_get_description instead [-Wdeprecated-declarations]
  407 |     const esp_app_desc_t* app_desc = esp_ota_get_app_description();
      |     ^~~~~
In file included from lib/bluepad32_arduino/arduino_platform.c:16:
/Users/hans/.platformio/packages/framework-espidf/components/app_update/include/esp_ota_ops.h:53:23: note: declared here
   53 | const esp_app_desc_t *esp_ota_get_app_description(void) __attribute__((deprecated("Please use esp_app_get_description instead")));
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Archiving .pio/build/esp32-s3-devkitc-1/lib386/libbluepad32_arduino.a
Indexing .pio/build/esp32-s3-devkitc-1/lib386/libbluepad32_arduino.a
Linking .pio/build/esp32-s3-devkitc-1/firmware.elf
Retrieving maximum program size .pio/build/esp32-s3-devkitc-1/firmware.elf
Checking size .pio/build/esp32-s3-devkitc-1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  25.9% (used 85008 bytes from 327680 bytes)
Flash: [=         ]   8.4% (used 702481 bytes from 8388608 bytes)
Building .pio/build/esp32-s3-devkitc-1/firmware.bin
esptool.py v4.7.5
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
======================================= [SUCCESS] Took 66.87 seconds =======================================
```

closed 2024-07-27 by Jason2866

ricardoquesada, 2024-07-28:
Of course, thanks!

Using `platform = https://github.com/pioarduino/platform-espressif32.git#develop` fixed the issue.

Many thanks!

MCUdude, 2024-07-31:
Could one of you (@ricardoquesada or @Jason2866) share a a compolete project that will compile?

What I'm trying to accomplish is to use "Arduino as component" inside IDF 5.x. The thing is that I got an [ESP32-S3 with a 7" touch screen](https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-7-inch.html), and I need espidf 5+ in order to get the performence I need (double buffers in PSRAM, etc.).  I got it up and running just fine, but I have a bunch of Arduino libraries I want to use with it as well. I've tried a bunch of things, but I'm always getting the following error:

(BTW this is the output when building the `esp-idf-arduino-bluepad32-template` from the development branch that @ricardoquesada provided:

```
Executing task: C:\Users\Hans\.platformio\penv\Scripts\platformio.exe run --environment esp32 

Processing esp32 (board: esp32dev; platform: https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip; framework: arduino, espidf)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (51.3.3) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.3 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.dc859c1e67 
 - framework-espidf @ 3.50104.0 (5.1.4) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.7.5 
 - tool-idf @ 1.0.1 
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.9.0 
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823 
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
No pyvenv.cfg file
CalledProcessError: Command '['C:\\Users\\Hans\\.platformio\\penv\\.espidf-5.1.4\\Scripts\\python.exe', '-m', 'pip', 'list', '--format=json', '--disable-pip-version-check']' returned non-zero exit status 106.:
  File "C:\Users\Hans\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Hans\.platformio\platforms\espressif32@src-d08281ff178025e4839b53f6d71eb00b\builder\main.py", line 363:
    target_elf = env.BuildProgram()
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hans\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hans\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hans\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 676:
    return method(*args, **kw)
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Hans\.platformio\platforms\espressif32@src-d08281ff178025e4839b53f6d71eb00b\builder\frameworks\espidf.py", line 1334:
    install_python_deps()
  File "C:\Users\Hans\.platformio\platforms\espressif32@src-d08281ff178025e4839b53f6d71eb00b\builder\frameworks\espidf.py", line 1212:
    installed_packages = _get_installed_pip_packages(python_exe_path)
  File "C:\Users\Hans\.platformio\platforms\espressif32@src-d08281ff178025e4839b53f6d71eb00b\builder\frameworks\espidf.py", line 1171:
    pip_output = subprocess.check_output(
  File "C:\Users\Hans\.platformio\python3\Lib\subprocess.py", line 466:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\Hans\.platformio\python3\Lib\subprocess.py", line 571:
    raise CalledProcessError(retcode, process.args,
====================================================================================== [FAILED] Took 1.25 seconds ======================================================================================

Environment    Status    Duration
-------------  --------  ------------
[... cut ...]
================================================================================ 1 failed, 0 succeeded in 00:00:01.254 ================================================================================ 

 *  The terminal process "C:\Users\Hans\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

Any idea how I can get a bare minimum espidf v5.x up and running while still being able to use "Arduino as component"?

Jason2866, 2024-07-31:
You find an example here https://github.com/pioarduino/platform-espressif32/tree/develop/examples/espidf-arduino-blink

MCUdude, 2024-08-01:
@Jason2866 thanks for the hint! I've tried the pioarduino `espidf-arduino-blink` example, but I'm getting the same error as Ive gotten with other "arduino as component" projects:

```
Executing task: C:\Users\Hans\.platformio\penv\Scripts\platformio.exe run 

Processing esp32-s3 (board: 4d_systems_esp32s3_gen4_r8n16; platform: espressif32; framework: arduino, espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/4d_systems_esp32s3_gen4_r8n16.html
PLATFORM: Espressif 32 (51.3.3+develop.sha.18cb216) > 4D Systems GEN4-ESP32 16MB (ESP32S3-R8N16)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.3 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.5a26d8ae8e 
 - framework-espidf @ 3.50104.0 (5.1.4) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.7.5 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.9.0 
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823 
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
No pyvenv.cfg file
CalledProcessError: Command '['C:\\Users\\Hans\\.platformio\\penv\\.espidf-5.1.4\\Scripts\\python.exe', '-m', 'pip', 'list', '--format=json', '--disable-pip-version-check']' returned non-zero exit status 106.:
  File "C:\Users\Hans\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Hans\.platformio\platforms\espressif32\builder\main.py", line 363:
    target_elf = env.BuildProgram()
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hans\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hans\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hans\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 676:
    return method(*args, **kw)
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Hans\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Hans\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 1385:
    install_python_deps()
  File "C:\Users\Hans\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 1263:
    installed_packages = _get_installed_pip_packages(python_exe_path)
  File "C:\Users\Hans\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 1222:
    pip_output = subprocess.check_output(
  File "C:\Users\Hans\.platformio\python3\Lib\subprocess.py", line 466:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\Hans\.platformio\python3\Lib\subprocess.py", line 571:
    raise CalledProcessError(retcode, process.args,
```

I have no idea how to resolve this, and I'm not able to run `pio run -t menuconfig`, because the project doesn't build

Jason2866, 2024-08-01:
The Platformio setup is broken `No pyvenv.cfg file` Close VSC.  Delete the complete hidden folder `.platformio`. Delete all generated folders from VSC and Platformio in the project folder. Start VSC, wait and follow the commands what to do.

TD-er, 2024-08-01:
I know that Python 3.12 + PlatformIO + Windows also may appear to be broken.
No idea whether that was some library which may have been updated since I tried, but I know that reverting back to Python 3.11 also helped a few users.

MCUdude, 2024-08-02:
Thank you @Jason2866! removing the .platoformio folder did the trick!