pioarduino/platform-espressif32 #14: Project build fails - missing partitions-4MB-tinyuf2.csv

davexre, 2024-07-30:
Just trying this project for the first time (hoping to overcome an issue I was seeing with an Adafruit ESP32-S3 Reverse TFT board). I switched my platformio.ini file to use the current stable release of this project, and see this when I go to build:

```
Compiling .pio/build/default/FrameworkArduinoVariant/variant.cpp.o
Compiling .pio/build/default/src/main.cpp.o
*** [.pio/build/default/partitions.bin] Source `partitions-4MB-tinyuf2.csv' not found, needed by target `.pio/build/default/partitions.bin'.
====================================================================================== [FAILED] Took 2.93 seconds ======================================================================================
```

Missing dependency, maybe?

Jason2866, 2024-07-30:
Post the complete `platformio.ini` to help.
I don't understand why using tinyuf2 in general. Eats up flash space...

davexre, 2024-07-30:
`platformio.ini` is:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:default]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.03/platform-espressif32.zip
board = adafruit_feather_esp32s3_reversetft
framework = arduino
lib_deps = 
	adafruit/Adafruit GFX Library@^1.11.10
	adafruit/Adafruit ST7735 and ST7789 Library@^1.10.4
	adafruit/Adafruit BusIO@^1.16.1
	adafruit/Adafruit NeoPixel@^1.12.3
	adafruit/Adafruit seesaw Library@^1.7.8
	Wire
	SPI
monitor_port = /dev/cu.usbmodem14401
monitor_speed = 115200
build_flags = -D GEM_DISABLE_U8G2
```

Looks like the partition file is being referenced from here: https://github.com/pioarduino/platform-espressif32/blob/77b00e457150251035218c89fbee064c902041dc/boards/adafruit_feather_esp32s3_reversetft.json#L5

mathieucarbou, 2024-07-30:
Maybe try a workaround and force set a partition file in your platform.io file:

`board_build.partitions = partitions-4MB-tinyuf2.csv`

and this file at the project root: `partitions-4MB-tinyuf2.csv` with [this content](https://github.com/espressif/arduino-esp32/blob/c4fd3838116b59989879485577b930d7579bc1cd/variants/adafruit_feather_esp32s3_reversetft/partitions-4MB-tinyuf2.csv):

```csv
# ESP-IDF Partition Table
 # Name,   Type, SubType, Offset,  Size, Flags
 # bootloader.bin,,          0x1000, 32K
 # partition table,          0x8000, 4K

 nvs,      data, nvs,      0x9000,  20K,
 otadata,  data, ota,      0xe000,  8K,
 ota_0,    0,    ota_0,   0x10000,  1408K,
 ota_1,    0,    ota_1,  0x170000,  1408K,
 uf2,      app,  factory,0x2d0000,  256K,
 ffat,     data, fat,    0x310000,  960K,
```

(content to be updated based on your needs)

Jason2866, 2024-07-30:
Fixed in branch develop. One of the many bullshit boards manifests provided and merged upstream.

closed 2024-07-30 by Jason2866