pioarduino/platform-espressif32 #20: Windows Paths not correct when building Adafruit Feather C6, cannot find ...-as.exe

tyeth, 2024-08-16:
Hey Jason,
Was trying to build the feather this week, which works fine in the Arduino IDE with arduino-esp32 v3.0.4.
Unforunately I can't get it to build in windows (but using WSL/linux worked fine) using the develop branch here nor the latest release with the feather board definition json file manually added, as it's constructing a non existant path when looking for some of the platform tools. I did try messing about with platform tool versions, but it felt something more fundamental.

The strange bit is that selecting the espressif C6 devkit-m (4mb c6) does build successfully using pioarduino, but not the adafruit feather c6.

Tested on two different machines, cleaned up ~/.platformio/packages and ./platforms first to check if leftover gubbins.

Error:
```
riscv32-esp-elf-g++: fatal error: cannot execute 'c:/users/tyeth/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
```

Jason2866, 2024-08-16:
Well, the windows path length limitation hits again. Probably because the boards name is quiet long.  Not really a issue from this fork. You can try following settings to shorten the path lengths used in the compile process.
```
; For unrelated compile errors with Windows it can help to shorten Platformio project path
workspace_dir = c:\.pio
core_dir = c:\.platformio
```

Jason2866, 2024-08-16:
If path names still to long you can try branch https://github.com/pioarduino/platform-espressif32/tree/poc_idf_installer This branch is more independent from Platformio registry and uses the IDF installer to install tools and toolchains. It uses shorter names for the toolchains to workaround the Windows path length limitation.
Use with:
```
platform = https://github.com/pioarduino/platform-espressif32.git#poc_idf_installer
```

tyeth, 2024-08-16:
Thanks Jason, I do wonder though as the executable name is wrong compared
to the ones available in my toolchain RISC-V folder, I wonder if it should
be riscv32-esp-elf-as.exe instead of as.exe

tyeth, 2024-08-16:
Here is the content of my folder:
[tree.txt](https://github.com/user-attachments/files/16638349/tree.txt)

Jason2866, 2024-08-16:
No, everything fine. If the shorten of the filenames I have suggested does not solve, I have no idea to solve. If you have a decent PC install WSL and compile with Linux. You will have shorter compile times using WSL too.

Jason2866, 2024-08-16:
Not every Windows limitation can be workarounded when using tools built and designed for Linux with Windows. The used gcc toolchain is coming from Linux...

Jason2866, 2024-08-16:
Closing since not an issue from pioarduino

closed 2024-08-16 by Jason2866