pioarduino/platform-espressif32 #21: Accessibility to Espressif and/or platformio

raphael-bmec-co, 2024-08-20:
I'd like to contribute here but I'm reluctant to do so if Espressif and/or platformio are going to be able to profit if/when they realise they've made a mistake - especially in light of the actions that lead to this point. Do the maintainers have a desire to prevent this from happening? If so what mechanisms are on hand?

raphael-bmec-co, 2024-08-20:
Oh @jason2866 I see it's you behind the wheel here (not surprised 😉). My question stands. What is the reciprocal relationship going to be with platformio and Espressif here?

Jason2866, 2024-08-20:
The answer is quiet simple. Platformio wanted more money and espressif didn't saw the ROI.
So there was no renewal of the contract. That's it. 
Platformio is Apache 2 licensed. espressif Arduino is under LGPL-2.1 license.
For the other part of the question, i answer like this. Join the Discord channel of pioarduino ;-)

Jason2866, 2024-08-20:
My hope for the future would be espressif and Platformio find a way to renew the partnership.
It would be a pleasure to see this work (when good enough and wanted ;-)) back in Platformio.
pioarduino is there because it is nice to have an actual espressif Arduino core

TD-er, 2024-08-20:
> there is no business relationship from pioarduino to any company!

Well....
We do mind our own business and we keep eachother company... ;)

closed 2024-08-22 by Jason2866