pioarduino/platform-espressif32 #22: Arduino as ESP-IDF component using this platform?

mairas, 2024-08-27:
Hi,

Thanks for the great initiative.

I'm trying to figure out whether it is possible and/or how to use [Arduino as an ESP-IDF component](https://community.platformio.org/t/arduino-as-espif-component-in-platformio-solution-that-works/36542) using this platform. My naïve solution of just using the following `platformio.ini` config didn't seem to work (the build process is identical to regular PlatformIO Arduino):

```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
board = esp32dev
framework = arduino, espidf
```

Jason2866, 2024-08-28:
See in examples how to do. It is working as exactly like this.

Jason2866, 2024-08-28:
Closing since no issue with pioarduino

closed 2024-08-28 by Jason2866