pioarduino/platform-espressif32 #24: requests.exceptions.JSONDecodeError: Extra data: line 1 column 4 (char 3)

sylque, 2024-08-28:
I don't know what happened one hour ago, but since then, I cannot get pioarduino to work. 

Each time I put this in my platformio.ini file...:
```ini
[env:debugNoBle]
platform = https://github.com/pioarduino/platform-espressif32
board = lolin_d32 # or whatever
framework = arduino
```
...I immediately get this platformio error:
```
Resolving debugNoBle dependencies...
JSONDecodeError: Traceback (most recent call last):
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\requests\models.py", line 974, in json
    return complexjson.loads(self.text, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\python3\Lib\json\__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\python3\Lib\json\decoder.py", line 340, in decode
    raise JSONDecodeError("Extra data", s, end)
json.decoder.JSONDecodeError: Extra data: line 1 column 4 (char 3)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\platformio\project\commands\init.py", line 106, in project_init_cmd
    install_project_dependencies(
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 107, in install_project_dependencies
    already_up_to_date = not install_project_env_dependencies(env, options)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 149, in _install_project_env_platform
    PlatformPackageManager().install(
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\platformio\package\manager\platform.py", line 60, in install
    p.configure_project_packages(project_env, project_targets)
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\platformio\platform\base.py", line 185, in configure_project_packages
    self.configure_default_packages(options, targets or [])
  File "C:\Users\Alex\.platformio\platforms\espressif32@src-923e3ebb70f703fbe384139b1453964b\platform.py", line 42, in configure_default_packages
    packjdata = requests.get(URL).json()
                ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alex\.platformio\penv\Lib\site-packages\requests\models.py", line 978, in json
    raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Extra data: line 1 column 4 (char 3)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```
Any idea?

qniens, 2024-08-28:
I encounter the same error, it happend after this pull request:
https://github.com/pioarduino/platform-espressif32/pull/23

Jason2866, 2024-08-28:
@qniens The PR #23 is only affecting branch `poc_idf_installer` When using 
```
platform = https://github.com/pioarduino/platform-espressif32
```
the branch `develop` is used.

Jason2866, 2024-08-28:
Why are you using branch `develop` in general? As described `develop` uses always latest Arduino master branch. It is normal that a develop branch sometimes is **not** working.
It is described in the readme how to use the stable release for Arduino Core 3.0.4

closed 2024-08-28 by Jason2866

qniens, 2024-08-28:
> @qniens The PR #23 is only affecting branch `poc_idf_installer` When using
> 
> ```
> platform = https://github.com/pioarduino/platform-espressif32
> ```
> 
> the branch `develop` is used.

Sure, here's a rewritten version of your comment:

---

You're right, I didn't pay enough attention. I'm familiar with using the stable release, but I was exploring the development branch to see if it would help with my current issue. I won't comment or report on the development branch anymore.

sylque, 2024-08-28:
@Jason2866 I chose to use the latest version to act as a beta tester and warn you about issues. Let me know if that's fine with you or if you prefer I use the stable version.

Jason2866, 2024-08-28:
@sylque Honestly i prefer issues only for stable version. Hints for all others please in Discord.
Reason my resources are limited and i try to solve issues as fast possible.
develop is not expected to be stable and working always. If issues are opened, I get a notice.
Not saying I do not try to solve the not stable branches. But this has a low priority.

It would be great to get an Info for not working "stuff" on Discord for all these.