pioarduino/platform-espressif32 #25: Some boards don't work: "pins_arduino.h: No such file or directory"

sylque, 2024-08-29:
At least two boards from M5 work well with platformio/espressif32, but not with pioarduino.

platformio.ini:
```ini
[env:m5stick-c]
;platform = platformio/espressif32 # -> WORKS
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip # -> DOESN'T WORKS
board = m5stick-c
framework = arduino
```
main.cpp:
```cpp
#include <Arduino.h>
void setup() {}
void loop() {}
```
Result:
```
In file included from C:/Users/Alex/.platformio/packages/framework-arduinoespressif32@src-0c9bc5a2e917d2a24b2fba29cb704cc7/cores/esp32/esp32-hal.h:75,
                 from C:/Users/Alex/.platformio/packages/framework-arduinoespressif32@src-0c9bc5a2e917d2a24b2fba29cb704cc7/cores/esp32/Arduino.h:36,
                 from src/main.cpp:1:
C:/Users/Alex/.platformio/packages/framework-arduinoespressif32@src-0c9bc5a2e917d2a24b2fba29cb704cc7/cores/esp32/esp32-hal-gpio.h:29:10: fatal error: pins_arduino.h: No such file or directory
```
Same issue with `board=m5stack-core-esp32`, but `board=m5stack-stamps3` works fine.

Jason2866, 2024-08-29:
@sylque Could you provide a PR for the missing/incorrect boards?

sylque, 2024-08-29:
See the above PR for fixing `board = m5stick-c`.

However, for `board=m5stack-stamps3`, there are 3 possible variants and I don't know the right one. So I let it for a future fix.

closed 2024-08-29 by Jason2866

Jason2866, 2024-08-29:
> However, for board=m5stack-stamps3, there are 3 possible variants and I don't know the right one. So I let it for a future fix.

That's why i said mess...