pioarduino/platform-espressif32 #27: Unable to Use Custom Libraries with PlatformIO

rvbc1, 2024-08-29:
**Description:**

I previously used the following configuration in my `platformio.ini` to include custom libraries:
```

[env]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.10+rc1/platform-espressif32.zip

platform_packages=
   framework-arduinoespressif32-libs @ https://github.com/rvbc1/esp32-arduino-lib-builder/releases/download/v3.1.0/esp32-arduino-libs-3.1.0.zip
```
   
This setup allowed me to use my custom SDK configuration, which is crucial for my project. However, this method is no longer working.

**Question:**

How can I fix this issue and continue using my custom libraries with PlatformIO? Is there an updated approach or alternative method to achieve the same result?

rvbc1, 2024-08-29:
Okay, I checked the official Espressif libraries, and they are working fine. I realized that I was using the master branch of the library builder instead of the v5.3 branch. I will check the v5.3 branch and provide an update.

rvbc1, 2024-08-29:
Okay, it was my mistake; everything is working now. However, I am getting the following warning for every compiled file:

```
Compiling build/hw_v1_debug/FrameworkArduino/main.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
```

Does anyone know the source of this problem? I didn’t add the ESP32 flag in the PlatformIO configuration.

Jason2866, 2024-08-29:
The CI examples do compile without any warnings.

closed 2024-08-29 by Jason2866