pioarduino/platform-espressif32 #30: Building with esp-matter as a managed component

HamzaHajeir, 2024-09-07:
Hi there,

I'm trying to build an empty sketch of esp-matter using arduino, esp-idf as frameworks, but failing with: 
<details>
<summary>pio run -v</summary>
```
Processing esp32-s3-devkitc-1 (platform: https://github.com/pioarduino/platform-espressif32.git#develop; board: esp32-s3-devkitc-1
; framework: espidf; board_build.embed_txtfiles:                                                                                  managed_components/espressif__esp_insights/server_certs/https_server.crt)
----------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (51.3.4+develop.sha.87d27ab) (git+https://github.com/pioarduino/platform-espressif32.git#develop) > Espress
if ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)                                                                                      HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, ol
imex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)                                           PACKAGES: 
 - framework-espidf @ 3.50104.0 (5.1.4, https://github.com/pioarduino/esp-idf/releases/download/v5.1.4.240805/esp-idf-v5.1.4.zip) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.7.5 (https://github.com/pioarduino/esptool/releases/download/v4.7.5/esptool.zip) 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Building ESP-IDF components for target esp32s3
Processing 13 dependencies:
[1/13] espressif/cbor (0.6.0~1)
[2/13] espressif/esp_diag_data_store (1.0.1)
[3/13] espressif/esp_diagnostics (1.0.2)
[4/13] espressif/esp_encrypted_img (2.1.0)
[5/13] espressif/esp_insights (1.0.1)
[6/13] espressif/esp_matter (1.3.0)
[7/13] espressif/esp_secure_cert_mgr (2.4.1)
[8/13] espressif/jsmn (1.1.0)
[9/13] espressif/json_generator (1.1.2)
[10/13] espressif/json_parser (1.0.3)
[11/13] espressif/mdns (1.4.0)
[12/13] espressif/rmaker_common (1.4.6)
[13/13] idf (5.1.4)
-- Found Python3: /root/.platformio/penv/.espidf-5.1.4/bin/python3 (found version "3.12.3") found components: Interpreter 
-- Project sdkconfig file /root/PlatformIO/Projects/pio-matter/sdkconfig.esp32-s3-devkitc-1
Loading defaults file /root/PlatformIO/Projects/pio-matter/sdkconfig.defaults...
-- Could NOT find Threads (missing: Threads_FOUND) 
-- App "pio-matter" version: 1.0
-- Adding linker script /root/PlatformIO/Projects/pio-matter/.pio/build/esp32-s3-devkitc-1/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /root/.platformio/packages/framework-espidf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /root/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /root/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /root/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /root/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /root/.platformio/packages/framework-espidf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /root/.platformio/packages/framework-espidf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: __pio_env app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_
app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__cbor espressif__esp_diag_data_store espressif__esp_diagnostics espressif__esp_encrypted_img espressif__esp_insights espressif__esp_matter espressif__esp_secure_cert_mgr espressif__jsmn espressif__json_generator espressif__json_parser espressif__mdns espressif__rmaker_common esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa             -- Component paths: /root/.platformio/packages/framework-espidf/components/__pio_env /root/.platformio/packages/framework-espidf/c
omponents/app_trace /root/.platformio/packages/framework-espidf/components/app_update /root/.platformio/packages/framework-espidf/components/bootloader /root/.platformio/packages/framework-espidf/components/bootloader_support /root/.platformio/packages/framework-espidf/components/bt /root/.platformio/packages/framework-espidf/components/cmock /root/.platformio/packages/framework-espidf/components/console /root/.platformio/packages/framework-espidf/components/cxx /root/.platformio/packages/framework-espidf/components/driver /root/.platformio/packages/framework-espidf/components/efuse /root/.platformio/packages/framework-espidf/components/esp-tls /root/.platformio/packages/framework-espidf/components/esp_adc /root/.platformio/packages/framework-espidf/components/esp_app_format /root/.platformio/packages/framework-espidf/components/esp_coex /root/.platformio/packages/framework-espidf/components/esp_common /root/.platformio/packages/framework-espidf/components/esp_eth /root/.platformio/packages/framework-espidf/components/esp_event /root/.platformio/packages/framework-espidf/components/esp_gdbstub /root/.platformio/packages/framework-espidf/components/esp_hid /root/.platformio/packages/framework-espidf/components/esp_http_client /root/.platformio/packages/framework-espidf/components/esp_http_server /root/.platformio/packages/framework-espidf/components/esp_https_ota /root/.platformio/packages/framework-espidf/components/esp_https_server /root/.platformio/packages/framework-espidf/components/esp_hw_support /root/.platformio/packages/framework-espidf/components/esp_lcd /root/.platformio/packages/framework-espidf/components/esp_local_ctrl /root/.platformio/packages/framework-espidf/components/esp_mm /root/.platformio/packages/framework-espidf/components/esp_netif /root/.platformio/packages/framework-espidf/components/esp_netif_stack /root/.platformio/packages/framework-espidf/components/esp_partition /root/.platformio/packages/framework-espidf/components/esp_phy /root/.platformio/packages/framework-espidf/components/esp_pm /root/.platformio/packages/framework-espidf/components/esp_psram /root/.platformio/packages/framework-espidf/components/esp_ringbuf /root/.platformio/packages/framework-espidf/components/esp_rom /root/.platformio/packages/framework-espidf/components/esp_system /root/.platformio/packages/framework-espidf/components/esp_timer /root/.platformio/packages/framework-espidf/components/esp_wifi /root/.platformio/packages/framework-espidf/components/espcoredump /root/PlatformIO/Projects/pio-matter/managed_components/espressif__cbor /root/PlatformIO/Projects/pio-matter/managed_components/espressif__esp_diag_data_store /root/PlatformIO/Projects/pio-matter/managed_components/espressif__esp_diagnostics /root/PlatformIO/Projects/pio-matter/managed_components/espressif__esp_encrypted_img /root/PlatformIO/Projects/pio-matter/managed_components/espressif__esp_insights /root/PlatformIO/Projects/pio-matter/managed_components/espressif__esp_matter /root/PlatformIO/Projects/pio-matter/managed_components/espressif__esp_secure_cert_mgr /root/PlatformIO/Projects/pio-matter/managed_components/espressif__jsmn /root/PlatformIO/Projects/pio-matter/managed_components/espressif__json_generator /root/PlatformIO/Projects/pio-matter/managed_components/espressif__json_parser /root/PlatformIO/Projects/pio-matter/managed_components/espressif__mdns /root/PlatformIO/Projects/pio-matter/managed_components/espressif__rmaker_common /root/. [... cut ...]
  managed_components/espressif__esp_insights/server_certs/https_server.crt
```

And some files and configurations added to meet up basic esp-matter projects in esp-idf.

### Reproduce
This repository holds a working project using the upstream platform: https://github.com/HamzaHajeir/pio-matter/

Changing platform to `https://github.com/pioarduino/platform-espressif32.git#develop` in platformio.ini is the needed edit.

This [issue](https://github.com/platformio/platform-espressif32/issues/936) could be similar.

Jason2866, 2024-09-08:
pioarduino branch develop uses IDF 5.1.4. Platformio develop uses IDF 5.3.x
So if you want to try with pioarduino you should do with pioarduino branch
https://github.com/pioarduino/platform-espressif32/tree/Arduino/IDF53

HamzaHajeir, 2024-09-08:
Thanks, 

I have tried that alongwith specifying Arduino as an ESP-IDF managed component pointing out to `idf-release/v5.3` branch instead of specifying arduino as a framework. At compiling I got the same error `sh: 1: cannot open lib/address_resolve/AddressResolve_DefaultImpl.h: No such file`.

Is it an issue in building script, or just versions mismatch?

Jason2866, 2024-09-09:
If this
```
platform = https://github.com/pioarduino/platform-espressif32.git#Arduino/IDF53
```
does not work i guess the newer IDF 5.3.1 version used in pioarduino is the reason for.

Jason2866, 2024-09-09:
fixed. Overlooked a small change to backport.