pioarduino/platform-espressif32 #34: Error compiling project

Basoil, 2024-09-22:
Compiling .pio/build/esp32s3wroom1/lib1c7/AsyncTCP_SSL/tcp_mbedtls.c.o
In file included from .pio/libdeps/esp32s3wroom1/AsyncTCP_SSL/src/tcp_mbedtls.c:33:
.pio/libdeps/esp32s3wroom1/AsyncTCP_SSL/src/tcp_mbedtls.h:35:10: fatal error: mbedtls/net.h: No such file or directory
   35 | #include "mbedtls/net.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.

Jason2866, 2024-09-22:
See Arduino migration guide https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

closed 2024-09-22 by Jason2866

BlueAndi, 2024-10-04:
@Jason2866 But the mbedTLS is currently not part, right?

Jason2866, 2024-10-04:
pioarduino uses orig. espressif Arduino release. So it is 100% same as espressif ArduinoIDE release.

BlueAndi, 2024-10-04:
Ah ... its moved to framework-arduinoespressif32-libs ... original it was in the framework-arduinoespressif32/tools/sdk/...