pioarduino/platform-espressif32 #35: Help some folks out by showing how to keep pioarduino in a special folder instead of the default espressif32 folder

trip5, 2024-10-08:
I figures out how to specify the version of espressif32 in my platoformio.ini keeps the "official" versions separate but it took me a bit too much time to figure out how to keep pioarduino in its own special folder.  I was mostly having problems because some projects I have (not mine) kept trying to use the pioarduino files which were residing in the espressif32 folder.

Would you mind dropping the following info somewhere on your front Readme and helping some people like me out?

---
The following specifies a specific folder to put the pioarduino files into instead of allowing it to conflict with any projects that may already be using the espressif32 folder:
````
platform = [name] = respository ; notes

platform = pioarduino51.03.05 = https://github.com/pioarduino/platform-espressif32.git#51.03.05 ; Arduino Core 3.0.5 / ESP-IDF v5.1.4+ (18 Sep 2024)
````

By the way, great work!  It's working great on an ESP32S3!

TD-er, 2024-10-08:
In your `platformio.ini` file, you can also define the folder to store your packages, cache and platforms folder, like this:
```
[platformio]
core_dir = .pioarduino
```
This will then store this folder in your project folder.
Make sure to add it to your `.gitignore` or else Git needs to handle a few 10'000 files :)

trip5, 2024-10-08:
That seems like a good solution if all else fails.

My solution isn't working anyway.  It downloaded correctly but can't build.

trip5, 2024-10-08:
I guess my problem was I was trying to work on 2 ESP projects at the same time, one of which relies on the official repo and another which relies on pioarduino (recoding an ESP32 project for ESP32-S3).  I think the solution offered by @TD-er is probably best for temporary solutions and the best solution is still to work with only one platform at a time and let platformio handle how the folders are operated on.  I see now they are being renamed as I open different projects which means the platform folder MUST be named espressif32 when it is in use.

closed 2024-10-08 by trip5

Xylopyrographer, 2024-10-08:
FWIW, in VS Code you can create separate Workspaces and enable/disable the PlatformIO and pioarduino extensions as needed.