pioarduino/platform-espressif32 #36: ESP-S3 - Debugger not starting

TheCrypt0, 2024-10-11:
Hey @Jason2866 ! Thanks for the great work porting the latest Arduino Core 3.0 to platformio, your project has been a literal game changer!

Unfortunately when trying to debug the sketch I get the following error when starting the debugger.

I completely removed the `~/.platformio` folder and cleaned the project. The directory `Users/user/.platformio/packages/tool-xtensa-esp-elf-gdb` exist but it is a folder, the permissions are set correctly.

I'm using the version `51.03.05`. Is there anything I should try?

Thanks again!

```
undefinedPermissionError: Traceback (most recent call last):
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/platformio/debug/cli.py", line 109, in cli
    _run(project_dir, debug_config, client_extra_args)
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/platformio/debug/cli.py", line 177, in _run
    loop.run_until_complete(coro)
  File "/Users/user/.platformio/python3/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/gdb.py", line 57, in run
    await self.spawn(*args, cwd=self.project_dir, wait_until_exit=True)
  File "/Users/user/.platformio/penv/lib/python3.11/site-packages/platformio/debug/process/base.py", line 76, in spawn
    await loop.subprocess_exec(
  File "/Users/user/.platformio/python3/lib/python3.11/asyncio/base_events.py", line 1694, in subprocess_exec
    transport = await self._make_subprocess_transport(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/python3/lib/python3.11/asyncio/unix_events.py", line 207, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/python3/lib/python3.11/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/Users/user/.platformio/python3/lib/python3.11/asyncio/unix_events.py", line 818, in _start
    self._proc = subprocess.Popen(
                 ^^^^^^^^^^^^^^^^^
  File "/Users/user/.platformio/python3/lib/python3.11/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Users/user/.platformio/python3/lib/python3.11/subprocess.py", line 1950, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: '/Users/user/.platformio/packages/tool-xtensa-esp-elf-gdb'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

Here's my `platformio.ini` file.

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nomad-v1]
# PlatformIO w/ Arduino Core 3 
# See: https://github.com/pioarduino/platform-espressif32/
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.05/platform-espressif32.zip
board = nomad-v1
framework = espidf, arduino

extra_scripts = 
    scripts/esp32_auto_flash.py
    scripts/esp32_merge_firmware.py

board_build.embed_txtfiles =
    managed_components/espressif__esp_insights/server_certs/https_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt

# Partition Table
board_build.partitions = partitions.csv

lib_deps =
    gitlab-friml/is31fl3733@^1.0.1
    hideakitai/DebugLog@^0.8.3
    SPI
    northernwidget/DS3231@^1.1.2
    bblanchon/ArduinoJson@^7.1.0

build_unflags = 
    -DARDUINO_USB_MODE=1
    -DCFG_TUSB_MCU
    -Os

build_flags =
    -O2
    -DARDUINO_USB_MODE=0
;    -DDEVELOPMENT_MODE

;monitor_port = /dev/cu.usbmodem0000000000001
monitor_speed = 115200
monitor_filters = direct
```

TheCrypt0, 2024-10-11:
Hey @Jason2866, weird.. I never had problems debugging with platformio with the ESP32-S3.

Btw, downgrading to `51.03.04` fixed the issue. It seems something related to the latest release.

Jason2866, 2024-10-12:
There is nothing changed in pioarduino from `51.03.04` to `51.03.05`

Jason2866, 2024-10-12:
@TheCrypt0 Please try following delete in hidden folder `.platformio` the folder `platforms` and in folder `packages` the folder `framework-arduinoespressif32` and try again

TheCrypt0, 2024-10-14:
I completely deleted the whole `~/.platformio` folder. Only downgrading to the version cited before fixed the issue, maybe it is something Arduino/ESP-IDF related?

Jason2866, 2024-10-19:
@TheCrypt0 I have no idea

closed 2024-11-03 by Jason2866