pioarduino/platform-espressif32 #41: Build error on 53.03.10-rc2 for C3 and C6

Laxilef, 2024-10-26:
Hi,

C3:
```
...
Compiling .pio\build\c3_mini\libd08\NimBLE-Arduino\nimble\nimble\host\src\ble_hs_id.c.o
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/Yurii/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory     
compilation terminated.
Compiling .pio\build\c3_mini\libd08\NimBLE-Arduino\nimble\nimble\host\src\ble_hs_log.c.o
```

C6:
```
...
Compiling .pio\build\esp32_c6\lib184\WiFi\WiFiSTA.cpp.o
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/Yurii/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory     
compilation terminated.
```

Perhaps builder cannot find this file: ``C:\Users\Yurii\.platformio\packages\toolchain-riscv32-esp\bin\riscv32-esp-elf-as.exe``

TD-er, 2024-10-26:
These are Windows issues, where the path length is too long.

By far the simplest way is to move the `.platformio` folder into your project dir.
```
[platformio]
core_dir = .platformio
```

Or maybe even a shorter version (`.pio`)

Make sure to add this folder also to your `.gitignore` or else you will be seeing a lot of files which should not be tracked.

Laxilef, 2024-10-27:
Strange, it compiled fine on RC1.
But your advice worked, thanks!

closed 2024-10-27 by Laxilef