pioarduino/platform-espressif32 #42: What do I set `platform` to if I want to always have the latest stable version?

puterboy, 2024-10-29:
Will setting:
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
```
ensure that platformio will always use the current stable release or do I need to update the above URL over time  to some version other than "51.03.07"
Alternatively, will something like
```
platform = https://github.com/pioarduino/platform-espressif32.git#main
```
ensure that platformio always uses the latest stable (non-development) release?

Thanks

closed 2024-11-03 by Jason2866

Jason2866, 2024-11-06:
There is no release of Arduino core based on IDF 5.1.5. So there is no Platform release for this combination.
You can combine stable Arduino core 3.0.7 with IDF 5.1.5 with this setup
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
platform_packages = framework-espidf @ https://github.com/pioarduino/esp-idf/releases/download/v5.1.5/esp-idf-v5.1.5.zip
```

Jason2866, 2024-11-06:
To be clear pioarduino is made to provide to use actual Arduino.
For this use case always the matching IDF version where the Arduino libs are built are provided. It is **not** the goal to provide stable IDF releases. I may do this, but there is no reason for.