pioarduino/platform-espressif32 #43: please add new boards definitions added in arduino-esp32 v3

mess-maker, 2024-11-01:
hi, and thanks for this awesome work

It seems that the pioarduino platform has the same board coverage as the offically supported pio ones based on v2.x.x

should it be possible to add the boards added with the realease of v3.x

as an example wt32-sc01-plus exist in the espressif github,
https://github.com/espressif/arduino-esp32/tree/master/variants/wt32-sc01-plus
this board is available in arduino ide but seems not present in pioarduino

the wt32-sc01-plus.json seems to be missing
regards

Jason2866, 2024-11-03:
See README. Needs to provided from the community

Mis1eader-dev, 2024-11-12:
I replaced platformio with pioarduino in hopes it will automatically upgrade everything, seems we have to manually include the URL of platform-espressif32 as per the README file
```ini
[env:stable]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
board = ...
...
```

Or go to pioarduino > Platforms > Advanced Installation, and put the above entry:
```
https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
```
Make sure to delete the old Espressif 32 platform, at the time of writing this comment it is 6.9.0, and upon completing the advanced installation it will add 51.03.07

Jason2866, 2024-11-12:
@Mis1eader-dev your post is off topic

Jason2866, 2024-11-17:
No boards manifest is auto generated in Platformio / pioarduino. As already mentioned if you want to add / change a boards manifest provide a PR.

mess-maker, 2024-11-17:
why be that rude ?
where did I say I don't like pio ?
I just asked some questions, and provided the missing file for this board
I just say I'm a beginner and I'd like someone review this file to confirm there no mistake in it before trying to go further (I don't know how to do a PR)

Jason2866, 2024-11-17:
Not being rude, just clear in communication. If you want that your suggestions is reviewed please learn how to submit a PR. That's the way to suggest changes. I don't have the time to do it different.

closed 2024-11-17 by Jason2866