pioarduino/platform-espressif32 #48: Where is the "Sponsor" button?

fredlcore, 2024-11-10:
Thank you so much for this. This repo saves me a lot of work and I'd like to support it, but can't find the "Sponsor" button.

Jason2866, 2024-11-10:
Thx a lot for the friendly words. Didn't think about a sponsor button. So you can't find one ;-)

fredlcore, 2024-11-10:
If you happen to come up with one, let people know :)...
Anyway, rest assured that you are doing the community a great service!

qniens, 2024-11-21:
Hi @Jason2866 , if you put up the _Buy me a coffee_ button  I'd be glad to treat you to one! 😊

kylefmohr, 2024-12-22:
@Jason2866 That's very generous of you. We all greatly appreciate all of your hard work on this. Is there any charitable cause or organization you'd want people to support on your behalf?

Jason2866, 2024-12-22:
@kylefmohr Well i am sure everyone knows an organization which is doing great.
So if you spend a little bit there, for honoring work people do for others i am happy, knowing that my work is maybe helping somewhere.

Jason2866, 2024-12-22:
Thank you @kylefmohr for the idea. Changed the readme accordingly.

closed 2024-12-22 by Jason2866