pioarduino/platform-espressif32 #49: Wire1 was not declared in this scope

troglodisme, 2024-11-11:
I have been using the below Tasmota platform in order to get my esp32-c6 to work

platform = https://github.com/tasmota/platform-espressif32/releases/download/2024.08.10/platform-espressif32.zip

I wanted to upgrade to pioarduino in order to use BLE but a lot of my existing libraries won't work as Wire1 seems to not be supported when using the pioarduino platformio?

Can anyone advise? 

`pio/libdeps/platformio/M5GFX/src/lgfx/v1/platforms/esp32/common.cpp: In member function 'void lgfx::v1::i2c::i2c_context_t::setPins(i2c_dev_t*, gpio_num_t, gpio_num_t)':
.pio/libdeps/platformio/M5GFX/src/lgfx/v1/platforms/esp32/common.cpp:794:51: error: 'Wire1' was not declared in this scope; did you mean 'Wire'?
  794 |         auto twowire = ((dev == &I2C0) ? &Wire : &Wire1);
      |                                                   ^~~~~
      |                                                   Wire
.pio/libdeps/platformio/M5GFX/src/lgfx/v1/platforms/esp32/common.cpp: In function 'cpp::bitwizeshift::result<void, lgfx::v1::error_t> lgfx::v1::i2c::release(int)':
.pio/libdeps/platformio/M5GFX/src/lgfx/v1/platforms/esp32/common.cpp:1003:44: error: 'Wire1' was not declared in this scope; did you mean 'Wire'?
 1003 |         auto twowire = ((i2c_port == 1) ? &Wire1 : &Wire);
      |                                            ^~~~~
      |                                            Wire
.pio/libdeps/platformio/M5GFX/src/lgfx/v1/platforms/esp32/common.cpp: In function 'cpp::bitwizeshift::result<void, lgfx::v1::error_t> lgfx::v1::i2c::setPins(int, int, int)':
.pio/libdeps/platformio/M5GFX/src/lgfx/v1/platforms/esp32/common.cpp:1049:42: error: 'Wire1' was not declared in this scope; did you mean 'Wire'?
 1049 |       auto twowire = ((i2c_port == 1) ? &Wire1 : &Wire);
      |                                          ^~~~~
      |                                          Wire
.pio/libdeps/platformio/M5GFX/src/lgfx/v1/platforms/esp32/common.cpp: In function 'cpp::bitwizeshift::result<void, lgfx::v1::error_t> lgfx::v1::i2c::init(int)':
.pio/libdeps/platformio/M5GFX/src/lgfx/v1/platforms/esp32/common.cpp:1071:42: error: 'Wire1' was not declared in this scope; did you mean 'Wire'?
 1071 |       auto twowire = ((i2c_port == 1) ? &Wire1 : &Wire);
      |                                          ^~~~~
      |                                          Wire`

TD-er, 2024-11-11:
Maybe this should include something like `soc_caps.h` ?
This file defines stuff like `SOC_HP_I2C_NUM` which is wrapped around the declaration of `Wire1`.

sivar2311, 2024-11-12:
The LovyanGFX library must respect that the ESP32-C series only has one I2C interface (See https://github.com/espressif/arduino-esp32/pull/10408)

Please file an issue to the LovyanGFX library.

ambient-works, 2024-11-12:
Thanks @sivar2311 the library worked completely fine before switching to pioarduino latest platform. 

This happens on many Adafruit libraries as well, not just M5GFX. 

Did you test on your end? Thank you!

troglodisme, 2024-11-12:
Thanks @sivar2311  the library worked completely fine before switching to pioarduino latest platform. 

This happens on many Adafruit libraries as well, not just M5GFX. 

Did you test on your end? Thank you!

TD-er, 2024-11-12:
There has been a recent change in the Arduino code which does actually set the correct number of I2C busses for the C6.
So "worked before" was more like "compiled and didn't crash before as I didn't try to access I2C bus 2". :)

closed 2024-11-13 by Jason2866