pioarduino/platform-espressif32 #50: esp_zigbee_core.h: No such file or directory with ESP32-C6

sans-ltd, 2024-11-13:
Hello,
we try to get the 53.03.10-rc2 version working for an ESP32-C6. This currently fails on esp_zigbee_core.hesp_zigbee_core.h.

Our work around to head on was to add 
```
build_flags = 
    -I /home/gonzo/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/espressif__esp-zigbee-lib
    -I /home/gonzo/.platformio/packages/framework-arduinoespressif32-libs/esp32c6/include/espressif__esp-zboss-lib/include
```
But this leads into the following compilation issues:
```
/home/gonzo/.platformio/packages/framework-arduinoespressif32/libraries/Zigbee/src/ep/ZigbeeThermostat.cpp: In member function 'void ZigbeeThermostat::setTemperatureReporting(uint16_t, uint16_t, float)':
/home/gonzo/.platformio/packages/framework-arduinoespressif32/libraries/Zigbee/src/ep/ZigbeeThermostat.cpp:188:20: error: cannot convert 'esp_zb_zcl_cmd_direction_t' to 'esp_zb_zcl_report_direction_t' in initialization
  188 |       .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                    |
      |                    esp_zb_zcl_cmd_direction_t
/home/gonzo/.platformio/packages/framework-arduinoespressif32/libraries/Zigbee/src/ep/ZigbeeTempSensor.cpp: In member function 'void ZigbeeTempSensor::reportTemperature()':
/home/gonzo/.platformio/packages/framework-arduinoespressif32/libraries/Zigbee/src/ep/ZigbeeTempSensor.cpp:83:19: error: 'esp_zb_zcl_report_attr_cmd_t' {aka 'struct esp_zb_zcl_report_attr_cmd_s'} has no member named 'cluster_role'
   83 |   report_attr_cmd.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
      |                   ^~~~~~~~~~~~
*** [.pio/build/automotive-addons/ZigbeeThermostat.cpp.o] Error 1
*** [.pio/build/automotive-addons/ZigbeeTempSensor.cpp.o] Error 1
```

Any hints how to fix this? Disabling Zigbee for now would also be fully OK.

Jason2866, 2024-11-13:
First do expect issues with an Release Candidate version ;-)
The bug is fixed today (2024.11.13) upstream in espressif Arduino repo.
You can use the latest fixed version with:
```
platform = https://github.com/pioarduino/platform-espressif32.git#Arduino/IDF53
```

Jason2866, 2024-11-13:
Yes, and even better it would be where the real issue is ;-)

closed 2024-11-13 by Jason2866