pioarduino/platform-espressif32 #53: compile without IPv6 enabled (#10582)

InnuendoPi, 2024-11-15:
does any branch include _[fix(net): Allow to compile without IPv6 enabled (#10582)_ yet?

Jason2866, 2024-11-15:
Yes, with the Hybrid compile branch. There you can modify sdkconfig settings and the Arduino libs are compiled with this changed settings before the compile run of the Arduino sketch is done with this libs. Used with:
```
platform = https://github.com/pioarduino/platform-espressif32.git#Arduino/IDF53_work_hybrid

[env:noipv6]
board                   = <your board>
custom_sdkconfig        =
                          '# CONFIG_LWIP_IPV6 is not set'
                          '# CONFIG_LWIP_IPV6_AUTOCONFIG is not set'
                          '# CONFIG_LWIP_PPP_SUPPORT is not set'
```

See some examples here: https://github.com/arendst/Tasmota/pull/22474/files

InnuendoPi, 2024-11-15:
perfect - thx for your support!

Jason2866, 2024-11-15:
Will be happy to compile Brautomat32 with when the source code is available ;-)

closed 2024-11-17 by Jason2866