pioarduino/platform-espressif32 #54: Compilation errros related to FS.h

IB-AST, 2024-11-15:
Compilation stops with following error:
...libraries/FS/src/FS.h:49:17: error: expected '{' before ';' token

Function call _p() in FS.h expects two parameters, but just one is given:
  File(FileImplPtr p = FileImplPtr()) : _p(p) {
    _timeout = 0;
  }

Thanks a lot in advance for any hint or debug suggestions.

Jason2866, 2024-11-15:
Please post your Platformio.ini  I don't know which branch or version you are using.

IB-AST, 2024-11-15:
Thanks for your reply. I tried those two:

;platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip

platform = https://github.com/pioarduino/platform-espressif32.git#develop

Jason2866, 2024-11-15:
CI examples do compile without any errors. There are examples which uses FS.
Check your code.

IB-AST, 2024-11-16:
Thanks again, but I am not even using FS functionality. Looks like FS library gets involved when including ESP <WebServer.h>

Jason2866, 2024-11-16:
Provide an minimal sketch where this happens. Without no help is possible

Jason2866, 2024-11-24:
Closing since no feedback

closed 2024-11-24 by Jason2866