pioarduino/platform-espressif32 #57: Coexistence with PlatformIO Broken

Xylopyrographer, 2024-11-19:
Earlier versions of pioarduino could coexist with PlatformIO. As in, you could select one by right-clicking its badge in the VSCode activity bar and disable the one you didn't want to use.

Latest version of pioarduino causes errors if both extensions are loaded. So now to disable, you have to select the extensions badge, disable one or the other, then quit and restart VSCode.

For example, with PlatformIO loaded, enable pioarduino. The following is logged:
```
Runtime Status
Activation
Activated by onCommand:platformio-ide.showHome event: 1ms

Uncaught Errors (1)
 command 'platformio-ide.showHome' already exists

Messages (22)
 Cannot register 'platformio-ide.activateProjectOnTextEditorChange'. This property is already registered.
 Cannot register 'platformio-ide.autoOpenPlatformIOIniFile'. This property is already registered.
 Cannot register 'platformio-ide.autoCloseSerialMonitor'. This property is already registered.
 Cannot register 'platformio-ide.autoRebuildAutocompleteIndex'. This property is already registered.
 Cannot register 'platformio-ide.showEditorTitleShortcuts'. This property is already registered.
 Cannot register 'platformio-ide.autoPreloadEnvTasks'. This property is already registered.
 Cannot register 'platformio-ide.customPATH'. This property is already registered.
 Cannot register 'platformio-ide.reopenSerialMonitorDelay'. This property is already registered.
 Cannot register 'platformio-ide.useBuiltinPython'. This property is already registered.
 Cannot register 'platformio-ide.useBuiltinPIOCore'. This property is already registered.
 Cannot register 'platformio-ide.useDevelopmentPIOCore'. This property is already registered.
 Cannot register 'platformio-ide.disablePIOHomeStartup'. This property is already registered.
 Cannot register 'platformio-ide.pioHomeServerHttpHost'. This property is already registered.
 Cannot register 'platformio-ide.pioHomeServerHttpPort'. This property is already registered.
 Cannot register 'platformio-ide.customPyPiIndexUrl'. This property is already registered.
 Cannot register 'platformio-ide.toolbar'. This property is already registered.
 A view with id platformio-ide.projectTasks is already registered.
 A view with id platformio-ide.quickAccess is already registered.
 A view with id platformio-debug.peripherals is already registered.
 A view with id platformio-debug.registers is already registered.
 A view with id platformio-debug.memory is already registered.
 A view with id platformio-debug.disassembly is already registered.
```
 Now using the Extensions manager, disable PlatformIO and pioarduino and then enable pioarduino.  pioarduino then throws:
```
Error: Could not find compatible pioarduino Core. Please enable `pioarduino-ide.useBuiltinPIOCore` setting and restart IDE. at FA.install (/Users/xylo/.vscode/extensions/pioarduino.pioarduino-ide-1.0.4/node_modules/pioarduino-node-helpers/dist/index.js:1:976643) at y.install (/Users/xylo/.vscode/extensions/pioarduino.pioarduino-ide-1.0.4/dist/extension.js:1:9256) at async /Users/xylo/.vscode/extensions/pioarduino.pioarduino-ide-1.0.4/dist/extension.js:1:40903...
```

Have to quit and relaunch VSC to resolve.

Can this be fixed to make, as in earlier versions, switching between the two is simple as before? Much better when working on projects that use other MCU's or if arduino-esp32 cores earlier than v3.0 are needed as the workspace can be set to enable/disable PlatformIO or pioarduino as required.

Using:
VSCode
     Version: 1.95.2
    Commit: e8653663e8840adaf45af01eab5c627a5af81807

PlatformIO
      PlatformIO 6.1.16
      Home 3.4.4

pioarduinorduino
    https://github.com/espressif/arduino-esp32/releases/download/3.0.7/esp32-3.0.7.zip

Host OS
    macOS 15.1

Jason2866, 2024-11-19:
Honestly I didnt know that was possible. I have no idea what is the reason for.
The VS Code plugin is here https://github.com/pioarduino/pioarduino-vscode-ide
Asking for help since my Typescript skills are only based on stackoverflow ;-)

Jason2866, 2024-11-20:
As discussed in Discord. Support of coexistence is not needed.

closed 2024-11-20 by Jason2866