pioarduino/platform-espressif32 #64: Custom bootloader build ? Suggestions needed

martinroger, 2024-11-26:
I am trying to see if I can get a custom bootloader build that omits the boot code dumped on UART0. I am aware that bootstrapping PIN15 to GND on start would silence the bootloader, unfortunately I am not at liberty to to do so because I am using 15 for the SD_MMC. 
Is there anyway from pioarduino (or from some fork) to change the sdkconfig and get a "silent" bootloader ? 
Any suggestions welcome.

TD-er, 2024-11-27:
If I'm not mistaken, there are some efuses which you can burn to fix those bootstrap states.
Maybe also worth looking at in the tech ref. guide?

N.B. you should really be careful when flashing such binary with thuse efuse burning steps, so maybe best to have it as some intermediate step during deployment.

martinroger, 2024-11-27:
> If I'm not mistaken, there are some efuses which you can burn to fixate those bootstrap states. Maybe also worth looking at in the tech ref. guide?
> 
> N.B. you should really be careful when flashing such binary with thuse efuse burning steps, so maybe best to have it as some intermediate step during deployment.

Thanks for replying... good call on the tech ref ([RTFM folks](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html)), but this is actually where I am getting a bit confused. 
This is the UART blob I am trying to silence, that is typically there when powering on the board : 
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4604
ho 0 tail 12 room 4
load:0x40078000,len:15468
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3180
entry 0x400805b8
```
Going through the documentation, here are a few things I think I understood, but please do correct me : 
- The message comes from the ROM bootloader ([source](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html))
- The "normal" way to silence those messages is to drive low GPIO 15 on start. It works, but the problem is that it is also connected to SDMMC CMD, and is not re-assignable (and in my case I need the SD)
- I can't think of a simple hardware solution to strap that GPIO15 to GND only for a short term on boot and let it be pulled up high the rest of the time...
- The config macro related to this blob is `CONFIG_BOOTLOADER_LOG_LEVEL` and should be set to _CONFIG_BOOTLOADER_LOG_LEVEL_NONE_ (or does this only affect RAM bootloader, aka second stage bootloader ?)
- Digging through the [espefuse documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/efuse.html) I can print out a summary of the eFuse table, but I don't see anything relating to the initial ROM bootloader UART blob
- I saw that on ESP32-S2 or S3 it seemed to be possible via eFuses, unfortunately I am stuck with ESP32 because I _need_ the bluetooth classic for A2DP

So all in all, is it even feasible or am I mistaken in looking at the (RAM) bootloader to suppress this ?

patfelst, 2024-11-27:
Normally you'd put a 100nF cap to ground on the pin, as the MTDO pin is set to internal pull-up by default. This pulls the pin to GND for the required 1ms at boot. If you use this pin for a slow MMC signal like "CS" it should be ok. I'm not sure what speed the other MMC signals are at, but the cap might delay transitions too much.

You might be able to play with small cap values to pull the pin low at boot, but not damp the MMC signals too mucy.

martinroger, 2024-11-27:
> Normally you'd put a 100nF cap to ground on the pin, as the MTDO pin is set to internal pull-up by default. This pulls the pin to GND for the required 1ms at boot. If you use this pin for a slow MMC signal like "CS" it should be ok. If you're using this pin for one of the faster MMC signals, the cap might delay transitions too much.
> 
> You might be able to play with small cap values to pull the pin low at boot, but not damp the MMC signals too much.

I guess I'm off to buy a bunch of caps then and try it out...

Jason2866, 2024-11-28:
Use the branch https://github.com/pioarduino/platform-espressif32/tree/Arduino/IDF53_work_hybrid which is a work in progress branch where breaking changes without a notice can happen. There you can change sdkconfig settings for Arduino and the bootloader. Use with
```
platform = https://github.com/pioarduino/platform-espressif32.git#Arduino/IDF53_work_hybrid
```
and do the wanted sdkconfig changes like in this example https://github.com/pioarduino/platform-espressif32/blob/a1e5f2bf4f60fcd45fba8e55ee4274d281b43204/examples/arduino-blink/platformio.ini#L48-L73