pioarduino/platform-espressif32 #70: Issue after new instal

koficek1, 2024-12-04:
Hello. I have problem after instal step by step PIOArduino. When i instal to new vscode extension PIOarduino. After instal core i restart Vscode and when pio want to load home page it show me this error. 

%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.95.3
  PIO IDE: v1.0.4
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: TypeError: Traceback (most recent call last):
  File "C:\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  %23 pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\.platformio\penv\Lib\site-packages\platformio\home\cli.py", line 92, in cli
    run_server(
  File "C:\.platformio\penv\Lib\site-packages\platformio\home\run.py", line 66, in run_server
    if not os.path.isdir(contrib_dir):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: isdir: s should be string, bytes, os.PathLike or integer, not NoneType

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\Vojtěch Kofroň\.vscode\extensions\pioarduino.pioarduino-ide-1.0.4\node_modules\pioarduino-node-helpers\dist\index.js:1:10938
	at ChildProcess.d (c:\Users\Vojtěch Kofroň\.vscode\extensions\pioarduino.pioarduino-ide-1.0.4\node_modules\pioarduino-node-helpers\dist\index.js:1:4560)
	at ChildProcess.emit (node:events:519:28)
	at ChildProcess.emit (node:domain:488:12)
	at maybeClose (node:internal/child_process:1105:16)
	at ChildProcess._handle.onexit (node:internal/child_process:305:5)
  ```

Jason2866, 2024-12-04:
Need help here. Don't have a Windows PC. Can't test in a VM since a different CPU architecture (Apple silicon).

Which Python version is installed and used?

Jason2866, 2024-12-04:
Do you have the Platformio Plugin installed? If yes disable or uninstall it. If you need this plugin for other projects keep it and do **not** install the pioarduino VSC plugin.
To have both plugins active is not possible

koficek1, 2024-12-04:
I install new VScode on PC where wasnt instaled VS before. After install it, add pioarduino Plugin to VScode. After automatic install plugin VS write me restart VScode. Then i want open main page and this error. My python is 3.13.0 64bit

xsbear, 2024-12-08:
same error, Mac 15.1.1 , Python 3.13 or Python 3.9.6

Basoil, 2024-12-10:
same error, Mac 12.7.5, Python 3.11
if in project platformio.ini
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip

but if switch to 
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.05/platform-espressif32.zip
have no error.

Jason2866, 2024-12-10:
@Basoil This finding is great help! Need to investigate why this slightly changed way of "finding" and downloading the last stable version is not working when installing from scratch.

Jason2866, 2024-12-10:
Please can all other, having this issue, try the workaround from Basoil?

```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
```

xsbear, 2024-12-10:
As @koficek1 said, can't open PIO Home page, or create project , when click this button, the errors are shown.

Jason2866, 2024-12-16:
Should be solved now. pioarduino will provide only Arduino builds based on IDF 5.3.x and Arduino Core 3.1.x
Having now only one build to support makes it possible to provide in branch `main` the latest stable build. The latest development version is available in branch `develop`

In short issue should be solved now.

closed 2024-12-17 by Jason2866

xsbear, 2024-12-17:
> Should be solved. Now pioarduino provides only Arduino builds based on IDF 5.3.x and Arduino Core 3.1.x Having only one build to support makes it possible to provide in branch `main` the latest stable build. The latest development version is available in branch `develop`
> 
> In short issue should be solved now.

Should wait for the new version release of vsc ext? I just try delete .platformio dir, and reinstall ext, still same error.

Jason2866, 2024-12-17:
@xsbear Did pioarduino a fresh download? If the old cached version was used it will not work.

xsbear, 2024-12-17:
@Jason2866 
Yes, I uninstall ext and remove $HOME/.platformio dir, then install ext， when click open PlO Home, still show the error.
[image]

 It seems PIO Home relevant error：
```
File "/Users/didi/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/Users/didi/.platformio/penv/lib/python3.9/site-packages/platformio/home/cli.py", line 92, in cli
    run_server(
  File "/Users/didi/.platformio/penv/lib/python3.9/site-packages/platformio/home/run.py", line 66, in run_server
    if not os.path.isdir(contrib_dir):
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/genericpath.py", line 42, in isdir
    st = os.stat(s)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType
```
in `platformio/home/run.py` line 66:
```
def run_server(host, port, no_open, shutdown_timeout, home_url):
  65     contrib_dir = get_core_package_dir("contrib-piohome")
  66     if not os.path.isdir(contrib_dir):
  67         raise PlatformioException("Invalid path to PIO Home Contrib")
```

may be `contrib_dir =. get_core_package_dir("contrib-piohome")`  return none?

Jason2866, 2024-12-17:
Ahh, pio home that's possible. Please open an new issue "Pio home crash at initial install"

Jason2866, 2024-12-17:
Does it work when you open an existing Platformio project?

xsbear, 2024-12-17:
> Does it work when you open an existing Platformio / pioarduino project? You can try for testing to clone this one. https://github.com/Jason2866/matter_test

Yes, it works on existing project clone from your matter_test.

Jason2866, 2024-12-17:
@xsbear Thx for testing. So the problem is "Pio home"

koficek1, 2024-12-24:
> Does it work when you open an existing Platformio / pioarduino project? You can try for testing to clone this one. https://github.com/Jason2866/matter_test

Hello, I test today this project and end with this error 

Linking .pio\build\esp32dev\firmware.elf
C:/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot open map file 

C:/Users/Vojtěch Kofroň/Documents/PlatformIO/Projects/matter_test-main/.pio/build/esp32dev/firmware.map: No such file or directory    

collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1

TD-er, 2024-12-24:
You can also set the `core_dir` to some (short) path inside your repository.

Just make sure that path is also mentioned in your `.gitignore`.

Drawback is that it will take up extra space on your harddrive if you have multiple checkouts of your repository.
However this can also be a good thing if you need to build several versions of your project at the same time.

koficek1, 2024-12-24:
> This is caused from Windows path length limit. Install and use WSL to workaround this Windows only problem. You can try to shorten path as max possible. Never use spaces and special characters in path as you do.

I have long patches enabled in registry value.

Jason2866, 2024-12-24:
@koficek1  Does not fix the problem completely.

Jason2866, 2024-12-24:
In other words. The tools and toolchains are ported from the Unix / Linux Universe to the Microsoft OS.
This works more or less good. There are no hacks included to overcome Windows limitations.

sivar2311, 2024-12-24:
Maybe the issue is with the non standard ascii characters in the pathname? This is also a known PIO bug.

koficek1, 2024-12-24:
I have this problem only with pioarduino platform with original platform is compile done

Jason2866, 2024-12-24:
You can't compare since it is not possible to compile Arduino core 3.1.0 with orig Platformio
Core 3.1.0 uses other (newer) toolchains. You compare apples vs. pears

koficek1, 2024-12-24:
> You can't compare since it is not possible to compile Arduino core 3.1.0 with orig Platformio Core 3.1.0 uses other (newer) toolchains. You compare apples vs. pears Anyways you have got answers how to solve.

I tested same steps on other PC and compile is done. Problem is maybe non standart ASCII in patch. Now i try it in WSL remote ubuntu 24.04.

TD-er, 2024-12-24:
> I tested same steps on other PC and compile is done. Problem is maybe non standart ASCII in patch. Now i try it in WSL remote ubuntu 24.04.

Don't forget to install `python3-venv` using apt, or else VS-code will not be able to install the build environment.

Basoil, 2024-12-27:
I think my case is similar.
Fully clean install VSC + Pioarduino
project platformio.ini:

`platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip`

compilation and upload is going fine
But, run QuickAccess > PIO Home > any of (Library, Boards, Platform, etc) causes an error:

>   VSCode: 1.96.2
>   PIO IDE: v1.0.4
>   System: Darwin, 21.6.0, x64 (Mac OS 12.7.6)
> 
>   ```
>   Error: TypeError: Traceback (most recent call last):
>   File "/Users/basil/.platformio/penv/lib/python3.11/site-packages/platformio/__main__.py", line 103, in main
>     cli()  %23 pylint: disable=no-value-for-parameter
>     ^^^^^
>   File "/Users/basil/.platformio/penv/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
>     return self.main(*args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/Users/basil/.platformio/penv/lib/python3.11/site-packages/click/core.py", line 1078, in main
>     rv = self.invoke(ctx)
>          ^^^^^^^^^^^^^^^^
>   File "/Users/basil/.platformio/penv/lib/python3.11/site-packages/platformio/cli.py", line 85, in invoke
>     return super().invoke(ctx)
>            ^^^^^^^^^^^^^^^^^^^
>   File "/Users/basil/.platformio/penv/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/Users/basil/.platformio/penv/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/Users/basil/.platformio/penv/lib/python3.11/site-packages/click/core.py", line 783, in invoke
>     return __callback(*args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/Users/basil/.platformio/penv/lib/python3.11/site-packages/platformio/home/cli.py", line 92, in cli
>     run_server(
>   File "/Users/basil/.platformio/penv/lib/python3.11/site-packages/platformio/home/run.py", line 66, in run_server
>     if not os.path.isdir(contrib_dir):
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "<frozen genericpath>", line 42, in isdir
> TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType
> 

Switch back to other releases nothing changes - error is still out

Jason2866, 2024-12-27:
Locked conversation. There is an open issue for the known bug with crashing Pio Home