pioarduino/platform-espressif32 #71: Missing function WiFi._setStatus()?

flakmoppen, 2024-12-09:
Firstly, I'm new to all this, so I might be completely wrong in posting this here, or to even expect this function to still work.

After starting to use this repo, `WiFi._setStatus()` no longer works, and VS Code complains that:
*class "WiFiClass" has no member "_setStatus"*

I'd love to not use this function, but unfortunately I need it because of [this issue](https://github.com/espressif/arduino-esp32/issues/7095).

Other than that, this has been awesome. I was missing some functionality that came with a newer version of idf, and using this repo solved those issues.
Thanks :)

Jason2866, 2024-12-09:
Indeed wrong place. That's either a question or an issue in IDF, not caused from pioarduino.
May you ask here https://www.esp32.com/viewforum.php?f=2

closed 2024-12-09 by Jason2866

flakmoppen, 2024-12-10:
In that case I'm sorry for taking up your time.
And thanks for pointing me in the right direction.