pioarduino/platform-espressif32 #75: missing SConscript file with Arduino core 3.1.0 based on IDF 5.3.2.241210

Abasz, 2024-12-16:
With the new release that updates arduino core to 3.1.0 I get the following error when trying to compile

*** missing SConscript file '.platformio/packages/framework-arduinoespressif32/tools/platformio-build.py'
File ".platformio/platforms/espressif32/builder/frameworks/arduino.py", line 36, in <module>

Jason2866, 2024-12-16:
CI completed successfully. Seems like an issue in your local installation. To solve try:
Close VSC. Delete hidden folder `.platformio` Start VSC

hobslob, 2024-12-17:
> CI completed successfully. Seems like an issue in your local installation. To solve try: Close VSC. Delete hidden folder `.platformio` Start VSC

Thank you @Jason2866 Had the same error here and deleting the .platformio folder worked for me but it took several VSC restarts for everything to catch up.

closed 2024-12-17 by Jason2866

Abasz, 2024-12-17:
Thanks this put on the right track. Packages and platforms is sufficient to be deleted. 

One additional note, I think the `-std=gnu++17` build flag came back, which was not present in 3.0.7 (as that version was compiling against C++23 by default based on my experience)