pioarduino/platform-espressif32 #77: esptool fails after upgrading to Arduino core 3.1.0 based on IDF 5.3.2.241210

sharmstr, 2024-12-17:
Not sure if this is the correct spot.  But just trying to upgrade from

platform = https://github.com/pioarduino/platform-espressif32.git#Arduino/IDF53_rc3
PLATFORM: Espressif 32 (53.3.10-rc3+sha.992c9ff) > Adafruit MatrixPortal ESP32-S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.1.0
 - framework-arduinoespressif32-libs @ 5.3.0+sha.a0f798cfc4
 - framework-espidf @ 3.50301.0 (5.3.1)
 - tool-cmake @ 3.21.3
 - tool-esptoolpy @ 4.8.1
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 3.2.0
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0)
 - toolchain-riscv32-esp @ 13.2.0+20240530
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530

to

platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
PLATFORM: Espressif 32 (53.3.10) > Adafruit MatrixPortal ESP32-S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.1.0
 - framework-arduinoespressif32-libs @ 5.3.0+sha.083aad99cf
 - framework-espidf @ 3.50302.0 (5.3.2)
 - tool-cmake @ 3.21.3
 - tool-esptoolpy @ 4.8.5
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 3.2.0
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0)
 - toolchain-riscv32-esp @ 13.2.0+20240530
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530

The latest builds fine, but get the following error when trying to upload

```
Building in release mode
Retrieving maximum program size .pio\build\adafruit_matrixportal_esp32s3_dev\firmware.elf
Checking size .pio\build\adafruit_matrixportal_esp32s3_dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.4% (used 40668 bytes from 327680 bytes)
Flash: [====      ]  36.8% (used 1229556 bytes from 3342336 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: COM8
Forcing reset using 1200bps open/close on port COM8
Waiting for the new upload port...
Uploading .pio\build\adafruit_matrixportal_esp32s3_dev\firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'C:\\Users\\sharmstr\\Documents\\bitbucket-repos\\cbled-espidf\\variants\\tasmota\\variants/adafruit_matrixportal_esp32s3/tinyuf2.bin'
*** [upload] Error 2
===================================================== [FAILED] Took 19.95 seconds =====================================================

Environment                        Status    Duration
---------------------------------  --------  ------------
adafruit_matrixportal_esp32s3_dev  FAILED    00:00:19.952
================================================ 1 failed, 0 succeeded in 00:00:19.952 ================================================ 

 *  The terminal process "C:\Users\sharmstr\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'adafruit_matrixportal_esp32s3_dev'" terminated with exit code: 1. 
```

Jason2866, 2024-12-17:
Please close VSC and delete the folder `.platformio` with all in there. You have left overs from the Tasmota Platformio fork which is conflicting.  
The log shows `'C:\\Users\\sharmstr\\Documents\\bitbucket-repos\\cbled-espidf\\variants\\tasmota\\variants/adafruit_matrixportal_esp32s3/tinyuf2.bin'`
There is no folder `variants\Tasmota` in official espressif pioarduino.

Jason2866, 2024-12-17:
Your Platformio setup is not valid. Do not define platform in [env] and again in the boards env.
Try with a pioarduino example first.

sharmstr, 2024-12-18:
I have no idea what Im doing so hopefully this isnt a waste of your time.  I took your advice and uploaded the sample espidf-arduino-blink example.  No issues.  From there I started adding parts of my project.  It starts to fail when I added my partitions.csv file.

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x330000,
app1,     app,  ota_1,   0x340000,0x330000,
littlefs,   data, littlefs,  0x670000,0x180000,
coredump, data, coredump,0x7F0000,0x10000,
```

I did some digging and it fails when this section of espidf.py is run (ignore my *** FLASH_EXTRA_IMAGES *** print statement)

```
if ota_partition_params["size"] and ota_partition_params["offset"]:
    # Generate an empty image if OTA is enabled in partition table
    ota_partition_image = os.path.join("$BUILD_DIR", "ota_data_initial.bin")
    if "arduino" in env.subst("$PIOFRAMEWORK"):
        ota_partition_image = os.path.join(ARDUINO_FRAMEWORK_DIR, "tools", "partitions", "boot_app0.bin")
    else:
        generate_empty_partition_image(ota_partition_image, ota_partition_params["size"])

    env.Append(
        FLASH_EXTRA_IMAGES=[
            (
                board.get(
                    "upload.ota_partition_offset", ota_partition_params["offset"]
                ),
                ota_partition_image,
            )
        ]
    )
    print("*** FLASH_EXTRA_IMAGES ***")
    EXTRA_IMG_DIR = join(env.subst("$PROJECT_DIR"), "variants", "tasmota")
    env.Append(
        FLASH_EXTRA_IMAGES=[
            (offset, join(EXTRA_IMG_DIR, img)) for offset, img in board.get("upload.arduino.flash_extra_images", [])
        ]
    )
```

I also noticed EXTRA_IMG_DIR = join(env.subst("$PROJECT_DIR"), "variants", "tasmota") was recently added.  53.03.10-rc3 worked fine.

working platformio.ini

```
[env]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
framework = arduino, espidf
build_flags = 
	-DCONFIG_BLINK_GPIO=2
monitor_speed = 115200
board_build.embed_txtfiles = 
	managed_components/espressif__esp_insights/server_certs/https_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt

[env:adafruit_matrixportal_esp32s3]
board = adafruit_matrixportal_esp32s3
board_build.filesystem = littlefs
lib_deps = 
	https://github.com/joltwallet/esp_littlefs.git
	adafruit/Adafruit seesaw Library@^1.7.9
	adafruit/Adafruit BusIO@^1.16.2
build_flags = 
	-DCONFIG_BLINK_GPIO=2
	-DCORE_DEBUG_LEVEL=3
	-DARDUINO_USB_MODE=1
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DUSE_GFX_ROOT=1
```

failing platformio.ini  (only difference is board_build.partitions = boards/partition.csv

```
[env]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
framework = arduino, espidf
build_flags = 
	-DCONFIG_BLINK_GPIO=2
monitor_speed = 115200
board_build.embed_txtfiles = 
	managed_components/espressif__esp_insights/server_certs/https_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
	managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt

[env:adafruit_matrixportal_esp32s3]
board = adafruit_matrixportal_esp32s3
board_build.partitions = boards/partition.csv
board_build.filesystem = littlefs
lib_deps = 
	https://github.com/joltwallet/esp_littlefs.git
	adafruit/Adafruit seesaw Library@^1.7.9
	adafruit/Adafruit BusIO@^1.16.2
build_flags = 
	-DCONFIG_BLINK_GPIO=2
	-DCORE_DEBUG_LEVEL=3
	-DARDUINO_USB_MODE=1
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DUSE_GFX_ROOT=1
```

So, not sure if Im on to something or not, but thought I'd let you know.

Jason2866, 2024-12-19:
First: Never tested any boards.json manifests which uses tinyuf2.
Looks like it is a conflict tying to use an individual partitions.csv with an boards manifest which has the fixed! tinyuf2 partition scheme.

closed 2024-12-19 by Jason2866

Jason2866, 2024-12-19:
Solution, if you don't want to use tinyuf2 don't use this boards manifest!