pioarduino/platform-espressif32 #84: Arduino core 3.1.0 based on IDF 5.3.2.241210

MicSG-dev, 2024-12-19:
When trying to compile the [WebServer](https://github.com/arduino-libraries/Ethernet/blob/master/examples/WebServer/WebServer.ino) example from the [Ethernet](https://github.com/arduino-libraries/Ethernet) library with [Arduino core 3.1.0 based on IDF 5.3.2.241210](https://github.com/pioarduino/platform-espressif32/releases/tag/53.03.10), it results in the following compilation errors:
```
src/main.cpp: In function 'void loop()':
src/main.cpp:75:43: error: cannot allocate an object of abstract type 'EthernetClient'
   75 |   EthernetClient client = server.available();
      |                           ~~~~~~~~~~~~~~~~^~
In file included from src/main.cpp:22:
.pio/libdeps/esp32dev/Ethernet/src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':
  214 | class EthernetClient : public Client {
      |       ^~~~~~~~~~~~~~
In file included from C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:197,
                 from src/main.cpp:20:
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32/cores/esp32/Client.h:29:15: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |   virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32/cores/esp32/Client.h:31:15: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |   virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
src/main.cpp:75:18: error: cannot declare variable 'client' to be of abstract type 'EthernetClient'
   75 |   EthernetClient client = server.available();
      |                  ^~~~~~
Compiling .pio\build\esp32dev\FrameworkArduino\FirmwareMSC.cpp.o
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:40:16: error: invalid abstract return type for member function 'EthernetClient EthernetServer::available()'
   40 | EthernetClient EthernetServer::available()
      |                ^~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:22:
.pio/libdeps/esp32dev/Ethernet/src/Ethernet.h:214:7: note:   because the following virtual functions are pure within 'EthernetClient':
  214 | class EthernetClient : public Client {
      |       ^~~~~~~~~~~~~~
In file included from C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:197,
                 from .pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:21:
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32/cores/esp32/Client.h:29:15: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |   virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
C:/Users/MichelGalvao/.platformio/packages/framework-arduinoespressif32/cores/esp32/Client.h:31:15: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |   virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'EthernetClient EthernetServer::available()':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:47:54: error: invalid cast to abstract class type 'EthernetClient'
   47 |         if (!chip) return EthernetClient(MAX_SOCK_NUM);
      |                                                      ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:72:40: error: invalid cast to abstract class type 'EthernetClient'
   72 |         return EthernetClient(sockindex);
      |                                        ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: At global scope:
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:75:16: error: invalid abstract return type for member function 'EthernetClient EthernetServer::accept()'
   75 | EthernetClient EthernetServer::accept()
      |                ^~~~~~~~~~~~~~
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'EthernetClient EthernetServer::accept()':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:82:54: error: invalid cast to abstract class type 'EthernetClient'
   82 |         if (!chip) return EthernetClient(MAX_SOCK_NUM);
      |                                                      ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:104:40: error: invalid cast to abstract class type 'EthernetClient'
  104 |         return EthernetClient(sockindex);
      |                                        ^
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp: In member function 'virtual size_t EthernetServer::write(const uint8_t*, size_t)':
.pio/libdeps/esp32dev/Ethernet/src/EthernetServer.cpp:170:18: error: cannot allocate an object of abstract type 'EthernetClient'
  170 |         available();
      |         ~~~~~~~~~^~
*** [.pio\build\esp32dev\libe72\Ethernet\EthernetServer.cpp.o] Error 1
```

The platformio.ini file looks like this:
```
[env:esp32dev]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
board = esp32dev
framework = arduino
lib_deps =
    https://github.com/arduino-libraries/Ethernet.git
```

I tried [Arduino Core 3.0.7 / IDF v5.1.4+](https://github.com/pioarduino/platform-espressif32/releases/tag/51.03.07) and it didn't show any compilation or execution errors (after uploading) and it worked perfectly. The platformio.ini file looks like this:
```
[env:esp32dev]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
board = esp32dev
framework = arduino
lib_deps =
    https://github.com/arduino-libraries/Ethernet.git
```

Jason2866, 2024-12-19:
Adding this is wrong. Arduino core has everything included.

```
lib_deps =
    https://github.com/arduino-libraries/Ethernet.git
```
Anyways not a pioarduino issue in general

closed 2024-12-19 by Jason2866

MicSG-dev, 2024-12-19:
I think you are wrong.

When using the following .ini:
```
[env:esp32dev]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip
board = esp32dev
framework = arduino
```

Results in the following error:
```
src/main.cpp:22:10: fatal error: Ethernet.h: No such file or directory

******************************************************************
* Looking for Ethernet.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Ethernet.h"
* Web  > https://registry.platformio.org/search?q=header:Ethernet.h
*
******************************************************************

   22 | #include <Ethernet.h>
      |          ^~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
```

And PlatformIO's own library search instructs to place the library in lib_deps (see [here](https://registry.platformio.org/libraries/arduino-libraries/Ethernet/installation)).

Also, how is it not a problem with pioarduino? In one version the compilation does not result in any errors, and in another compilation several errors occur. So I think it does have something to do with pioarduino.

sivar2311, 2024-12-19:
@MicSG-dev You need to `#include <ETH.h>`
See https://github.com/espressif/arduino-esp32/tree/master/libraries/Ethernet/src

MicSG-dev, 2024-12-19:
> @MicSG-devVocê precisa `#include <ETH.h>` ver https://github.com/espressif/arduino-esp32/tree/master/libraries/Ethernet/src

Thanks for the suggestion, but I already used this library a few days ago with the ESP32 and the W5500 and there were always random disconnections (triggering the `ARDUINO_EVENT_ETH_DISCONNECTED` event). With the Ethernet.h library there are no such disconnections. That's why I would prefer to use Ethernet.h.

sivar2311, 2024-12-19:
But the library you mentioned is not for ESP32 based boards
See https://github.com/arduino-libraries/Ethernet/tree/master/docs

Jason2866, 2024-12-19:
Still wrong place for this. Compile does work as expected. pioarduino is **not** responsible for not working as expected ;-)

sivar2311, 2024-12-19:
@MicSG-dev If you have trouble with the Espressif Ethernet library the correct place for an issue would be
https://github.com/espressif/arduino-esp32/issues