pioarduino/platform-espressif32 #88: IPv6/Openthreads issue with C6

motamman, 2024-12-21:
Anytime I use wifi in a manner that requires the domain name or hostname to be resolved, such as http.begin(<doman/path/filename>), I get errors like this:

Checking resolution for domain: mizzen.zennora.sv //my debug
E (1776) OPENTHREAD: esp_openthread_task_switching_lock_acquire(38): Failed to acquire the lock because the mutex is not ready
E (1777) OPENTHREAD: esp_openthread_task_switching_lock_release(46): Failed to release the lock because the mutex is not ready
E (1788) OT_DNS64: Cannot find NAT64 prefix
[  1231][E][NetworkManager.cpp:136] hostByName(): DNS Failed for 'mizzen.zennora.sv' with error '-54'
Domain resolved to IP: 0.0.0.0 //my debug

I tried this on both an Xiao esp32-c6 and espressif esp32-c6 and changed the boards accordingly in the platform.ini.  Same behavior.

It all works if I swap out the domain name with the IP. Same behavior when connecting to a mqtt broker or if I try to access a local hostname: IP only.

Same code, different chip (Xiao esp32-S3,) no problems. My code was ported from use with Arduino IDE, where it worked as expected.

Jason2866, 2024-12-21:
github issues is not the place to discuss individual programming problems.

closed 2024-12-21 by Jason2866

motamman, 2024-12-21:
I think is a bug. I neglected to tag it that way.

Jason2866, 2024-12-21:
pioarduino does "just" compile the code. It has no impact in compiled resulting code.

motamman, 2024-12-21:
I appreciate your work. I am simply trying to communicate, as sloppily as it may be, that the C6 implementation has a problem resolving domain names. The S3 implementation is fine. I am *just* trying to help. I have not the skills to delve in and offer a solution. You may ignore or consider it as you wish.

And, most sincerely, thank you.

TD-er, 2024-12-21:
What @Jason2866 is trying to tell here is that PioArduino is a project to help building projects.
This sounds like it may be an issue either with the [Espressif/Arduino-ESP32](https://github.com/espressif/arduino-esp32) code or the ESP-IDF SDK.

My first step would be to open an issue about it on the Arduino-ESP32 repository.

Jason2866, 2024-12-21:
@motamman Sorry when sounding rude. Was not my intention. I have to much of such not related stuff and writing always long explanation is consuming my already spare time a lot.

motamman, 2024-12-22:
Should anyone else come here asking about this problem ... It is a known issue with the esp32-arduino-lib-builder. (https://github.com/espressif/esp32-arduino-lib-builder/pull/257) and was supposedly patched yesterday. I haven't been able to get the patch to work yet, tho I am sure that's more me than them. Thanks, gentlemen. Pax.

KenFellows, 2024-12-24:
I am having the same issue.  Is there a fix for it yet?