pioarduino/platform-espressif32 #94: Build fails with "as.exe" file not found

juepi, 2024-12-27:
I've tried tried several ways to compile a firmware for a ESP32-C6 ("Super Mini" ali board with 4MB flash), however it constantly fails with the following error:

```
Executing task: C:\Users\juerg\.platformio\penv\Scripts\platformio.exe run 

Processing ESP32C6 (framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip; board: esp32-c6-devkitm-1)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c6-devkitm-1.html
PLATFORM: Espressif 32 (53.3.10) > Espressif ESP32-C6-DevKitM-1
HARDWARE: ESP32C6 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.1.0 
 - framework-arduinoespressif32-libs @ 5.3.0+sha.083aad99cf 
 - tool-esptoolpy @ 4.8.5 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-riscv32-esp @ 13.2.0+20240530
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PubSubClient @ 2.8.0
|-- ArduinoOTA @ 3.1.0
|-- WiFi @ 3.1.0
Building in release mode
Compiling .pio\build\ESP32C6\src\common-functions.cpp.o
Compiling .pio\build\ESP32C6\src\main.cpp.o
Compiling .pio\build\ESP32C6\src\mqtt-subscriptions.cpp.o
Compiling .pio\build\ESP32C6\src\setup.cpp.o
Compiling .pio\build\ESP32C6\src\user_setup_loop.cpp.o
Building .pio\build\ESP32C6\bootloader.bin
Generating partitions .pio\build\ESP32C6\partitions.bin
Compiling .pio\build\ESP32C6\lib94a\PubSubClient\PubSubClient.cpp.o
esptool.py v4.8.5
Creating esp32c6 image...
Merged 2 ELF sections
Successfully created esp32c6 image.
Compiling .pio\build\ESP32C6\libab8\Network\NetworkClient.cpp.o
Compiling .pio\build\ESP32C6\libab8\Network\NetworkEvents.cpp.o
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/juerg/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\ESP32C6\src\mqtt-subscriptions.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/juerg/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\ESP32C6\lib94a\PubSubClient\PubSubClient.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/juerg/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\ESP32C6\libab8\Network\NetworkEvents.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/juerg/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\ESP32C6\src\user_setup_loop.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/juerg/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\ESP32C6\src\main.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/juerg/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\ESP32C6\src\common-functions.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/juerg/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\ESP32C6\src\setup.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/juerg/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\ESP32C6\libab8\Network\NetworkClient.cpp.o] Error 1
========================================================================== [FAILED] Took 12.58 seconds ==========================================================================

Environment    Status    Duration
-------------  --------  ------------
ESP32C6        FAILED    00:00:12.585
===================================================================== 1 failed, 0 succeeded in 00:00:12.585 =====================================================================

```
This is especially strange as i can find the file `as.exe` in the mentioned path.

My relevant parts of the `platformio.ini` file:
```
[env:ESP32C6]
framework = arduino
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
; Chip type ESP32-C6FH4 (4MB flash on-chip)
; board config file in projects "boards" directory!
board = esp32-c6-devkitm-1
; Use partition table with maximized app space (for 4MB flash); reduces SPIFFS partition to 132kB
board_build.partitions = min_spiffs.csv
```

I am using [this board file](https://github.com/pioarduino/platform-espressif32/blob/main/boards/esp32-c6-devkitm-1.json) in my local `boards` directory.
I've tried building with the original PlatformIO VSC extension installed as well as PIOarduino. Same result.

Any help highly appreciated.

Thanks,
Juergen

Jason2866, 2024-12-27:
Known limitation of Windows. Caused by Windows path length limit.
Workarounds can be found here https://github.com/arendst/Tasmota/discussions/21964

juepi, 2024-12-28:
Oh great, i thought in 2024 we're past such BS.. 🙄

Thanks for the hint, closing case.

closed 2024-12-28 by juepi

juepi, 2024-12-28:
In case anyone else stumbles across this issue, i was able to work around it with a relatively simple solution:

a.) Issue the following command in Windows terminal (not elevated, same user you run VSC, adopt to your needs):
`subst p: C:\Users\Your_Username\.platformio`

b.) add the following to your `[platformio]` section of `platformio.ini`:
`core_dir = p:\`

Compilation for my ESP32-C6 works now.

yours,
Juergen