pioarduino/platform-espressif32 #95: OneWire example and missing GPIO define

netmaniac, 2024-12-29:
Hi!
First thanks You for effort to make new IDF available for Platformio. I have ran on problems with OneWire library. I try to use it with Platformio CLI (or VCS  is required?): Simple platformio.ini:

```
[env:heltec_wifi_lora_32_V3]
platform = espressif32@6.4.0
;platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
board = heltec_wifi_lora_32_V3
framework = arduino
lib_deps =
    OneWire@2.3.8
```
And this example: https://github.com/PaulStoffregen/OneWire/blob/master/examples/DS18x20_Temperature/DS18x20_Temperature.ino
When compiled against old platform - it compiles. Changing to pioarduino:

`.pio/libdeps/heltec_wifi_lora_32_V3/OneWire/util/OneWire_direct_gpio.h:173:17: error: 'GPIO' was not declared in this scope
`
https://github.com/PaulStoffregen/OneWire/blob/dae0fca24d256adc0d06c0969e837b914dd48366/util/OneWire_direct_gpio.h#L174

I have added 
```
#define XSTR(x) STR(x)
#define STR(x) #x
#pragma message "IDF: " XSTR(IDF_VER)
```
at example start and it shows `IDF: "v5.3.2-174-g083aad99cf-dirty"` for pioarduino 53.03.10 and `v4.4.5` for old platform. But with Arduino IDE 2.x example compiles against the same IDF 5.3.2.

I don't know how this works :) GPIO object is defined in gpio_struct.h. Older frameworks have them in tools/sdk directory. pioarduino does not have tools/sdk at all. Arduino IDE have gpio_struct.h in `packages/esp32/tools/esp32-arduino-libs`

Comparing output of `pio pkg list` for old platform I see that pioarduino requires `framework-arduinoespressif32-libs @ 5.3.0+sha.083aad99cf`, there is gpio_struct.h but somehow is not being included?

sivar2311, 2024-12-29:
@netmaniac 
There are some differences to note in the ESP-IDF versions 4.4.x to 5.3.x as well as in Arduino 2.x to 3.x.

The following applies to ESP-IDF 5.3.x and Arduino 3.x: `#include “soc/gpio_struct.h”`.
See https://github.com/espressif/arduino-esp32/issues/8846

The OneWire library must take this into account when using this file. 
Therefore, pioarduino is the wrong place for you - pioarduino is a build system.
The repository of the [OneWire library is the right place to report your issue](https://github.com/PaulStoffregen/OneWire/issues). 
Please send an issue there.

netmaniac, 2024-12-29:
I'm not sure - shouldn't this platform with framework Arduino mimic behavior of Arduino IDE? Current Arduino IDE is based on Espressif ESP32 v 3.1.0 and OneWire library there does not need any tweaks. Shouldn't be same here?

sivar2311, 2024-12-29:
Does the OneWire library work under the ArduinoIDE without any problems?
It must be ensured that both the Arduino core and the OneWire library used are absolutely identical!

Again: pioarduino is only the build system and the error you observed was already mentioned in the espressif arduino repository.

sivar2311, 2024-12-29:
The [DS18x20_Temperature example](https://github.com/PaulStoffregen/OneWire/blob/master/examples/DS18x20_Temperature/DS18x20_Temperature.ino)  compiles fine without any issues:

`platformio.ini`

```
[env:esp32dev]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
board = esp32dev
framework = arduino
lib_deps = 
  https://github.com/PaulStoffregen/OneWire @ 2.3.8
```

closed 2024-12-29 by Jason2866

netmaniac, 2024-12-29:
Strange. On my developer account with plenty pio packages installed - it have started compiling. I have created new user (Ubuntu 22.04) and installed fresh Platformio CLI - it does not work. Will it work with CLI or need VCS and `pioarduino ide` extension? If it should with Platformio CLI I can try to create Vagrant script to demonstrate problem?

Jason2866, 2024-12-29:
VSC with pioarduino plugin is always optional with pioarduino.
There is no difference in compile process using CLI or VSC.
You can push your setup to your github account

netmaniac, 2024-12-29:
OK, Install [Vagrant](https://developer.hashicorp.com/vagrant/install) if not yet installed. Unpack [vagrant.zip](https://github.com/user-attachments/files/18270852/vagrant.zip) and run shell. Go to unpacked directory and run `vagrant up`. It will install Ubuntu image and install Platformio CLI on it. Then `vagrant ssh` and when connected to virtual machine: `cd /vagrant/pio-test && pio run`

Result - `error 'GPIO'.....`

Jason2866, 2025-01-01:
@netmaniac Well vagrant is not starting up
```
The VirtualBox VM was created with a user that doesn't match the
current user running Vagrant. VirtualBox requires that the same user
be used to manage the VM that was created. Please re-run Vagrant with
that user. This is not a Vagrant issue.

The UID used to create the VM was: 1000
Your UID is: 501
```

netmaniac, 2025-01-04:
@Jason2866 I'm sorry, I was sharing Vagrant project fot the first time. You can just remove ./vagrant/.vagrant directory or use attached zip with this directory already removed.
[vagrant2.zip](https://github.com/user-attachments/files/18308572/vagrant2.zip)

Jason2866, 2025-01-13:
@netmaniac  As already said. outdated One Wire lib. Use this settings

```python
[env:heltec_wifi_lora_32_V3]
board = esp32dev
;board = heltec_wifi_lora_32_V3
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
framework = arduino
lib_deps =
	https://github.com/PaulStoffregen/OneWire
```