pioarduino/platform-espressif32 #105: Does pioarduino ensure compatibility and stability?

Mis1eader-dev, 2025-01-14:
pioarduino seems like a great alternative to platformio, which still stays on arduino-esp32 2.x.
After giving it a try for the 2nd time, I still am unable to make it work with pioarduino, the ESP32 board resets upon boot:
```
ELF file SHA256: 35d5f4dbb

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4688
load:0x40078000,len:15448
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3196
entry 0x400805a4

assert failed: vApplicationGetTimerTaskMemory port_common.c:94 (pxTCBBufferTemp != NULL)

Backtrace: 0x4008337d:0x3ffe39c0 0x4008a5a1:0x3ffe39e0 0x40090172:0x3ffe3a00 0x4008b691:0x3ffe3b30 0x4008b9dc:0x3ffe3b50 0x4008d6ce:0x3ffe3b90 0x40129cb3:0x3ffe3be0 0x400e155a:0x3ffe3c10 0x4008288e:0x3ffe3c40 0x40079672:0x3ffe3c90 |<-CORRUPTED
```

I am unsure if this is the right place to create this issue in. The platformio.ini file uses:
```ini
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = esp32dev
framework = arduino
```

This could be an issue with ESP-IDF, but I am unaware of where the culprit is. Let me know if I should create an issue elsewhere.

Mis1eader-dev, 2025-01-14:
That's really sad :(

Jason2866, 2025-01-14:
Closing since it is not a pioarduino issue.

closed 2025-01-14 by Jason2866