pioarduino/platform-espressif32 #106: TinyUSB stack fails to come up on ESP32-S3 device

i-am-shodan, 2025-01-14:
When using `platform = https://github.com/platformio/platform-espressif32.git` I am able to use TinyUSB to create a USB serial device that appears on Windows. When I change my platform to use pioarduino `platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip` the USB stack doesn't come up. There is no error in Windows device manager.

Here is a very simple test case to reproduce. I'm using a LilyGo Dongle S3 which uses an ESP32-S3.

main.cpp
```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>

void setup() {
  TinyUSBDevice.begin(0);
  Serial.begin(115200);
}

void loop() {
  Serial.println("Loop!");
  Serial.flush();
  delay(1000);
}
```

platformio.ini
```
[env:esp-32-s3]
#platform = https://github.com/platformio/platform-espressif32.git # works
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip # does not work
board = esp32-s3-devkitc-1
framework = arduino
monitor_filters = esp32_exception_decoder
monitor_speed = 115200
build_unflags = 
	-DARDUINO_USB_MODE=1
build_flags = 
	-std=gnu++2a
	-DARDUINO_USB_MODE=0
	-DARDUINO_USB_CDC_ON_BOOT=1
	-Wl,-z,muldefs
	-D CFG_TUD_ENABLED
lib_deps = 
	adafruit/Adafruit TinyUSB Library@3.3.4
```

TD-er, 2025-01-14:
Probably related: https://github.com/espressif/arduino-esp32/issues/10836

I always use USB HWCDC except for the ESP32-S2.
TinyUSB isn't working very well. If you only need to add some serial port, then please use HWCDC where possible.

i-am-shodan, 2025-01-14:
Well I actually need USB HID, MSC and a load of others but I wanted to give you a tiny test case.

Jason2866, 2025-01-14:
Wrong github for this issue. pioarduino is "just" the build system. There is no change in Arduino Core.
Please open an issue in espressif32 Arduino github with an example that fails.

closed 2025-01-14 by Jason2866