pioarduino/platform-espressif32 #109: How to make use of esp32-arduino-lib-builder's output for pioarduino?

ceewanna, 2025-01-24:
1. How to make use of esp32-arduino-lib-builder's output for pioarduino? I saw the the content in .platformio/packages/framework-arduinoespressif32-libs is similar to esp32-arduino-lib-builder's output. Any means to make use of such in pioarduino?

2. As recommended in th frontage to use pioarduino ide, what are the differences between pioarduino ide and platformio ide? 

Does pioarduino have discord?

sivar2311, 2025-01-24:
> 2. Does pioarduino have discord?

See 2nd line in https://github.com/pioarduino/platform-espressif32/blob/main/README.md

[image]

ceewanna, 2025-01-24:
@sivar2311 Thanks. It's just too obvious, isn't it? :)

Jason2866, 2025-01-24:
Replace this entrys with yours
https://github.com/pioarduino/platform-espressif32/blob/5920b375a85c3f289bfa72cf7e79755b9315cf7d/platform.json#L36 and https://github.com/pioarduino/platform-espressif32/blob/5920b375a85c3f289bfa72cf7e79755b9315cf7d/platform.json#L42

ceewanna, 2025-01-24:
@Jason2866 
Both look the same to mine pioarduino 53.03.11. But the second which comes for framework-arduinoespressif32-libs looks like the structure of what has come out from esp32-arduino-lib-builder. 

From platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip
is it possible to config pioarduino with platform_packages =  framework-arduinoespressif32-libs @ <location of the lib builder output>

Jason2866, 2025-01-24:
The first zip is more or less a zip from the branch master tagged with v3.1.1. So if you did no changes to the Arduino Source code you don't have to change here anything.
Yes, the second one is the output of the lib Builder. You can specify exactly as you wrote.
If the file is locally on your PC the entry is.
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip
platform_packages = framework-arduinoespressif32-libs @ file:///<path to your arduuino builder.zip>
```

ceewanna, 2025-01-24:
@Jason2866 Wow it works nicely with this approach.

closed 2025-01-25 by Jason2866