pioarduino/platform-espressif32 #110: Compilation Issue: FileNotFoundError with missing cache pkg file.

lonerzzz, 2025-01-24:
Hi I am trying to compile, but get the following error right out of the box:

```
Processing esp-wrover-kit (platform: https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip; board: esp-wrover-kit; framework: arduino)
--------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.3/esp32-arduino-libs-idf-release_v5.3-cfea4f7c-v1.zip
Unpacking 0%Unpacking...
FileNotFoundError: Traceback (most recent call last):
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\manager\_install.py", line 37, in unpack
    return fu.unpack(dst, with_progress=with_progress)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\unpack.py", line 186, in unpack
    self._archiver.extract_item(item, dest_dir)
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\unpack.py", line 49, in extract_item
    self._afo.extract(item, dest_dir)
  File "E:\Program Files\Programming\Python\Lib\zipfile\__init__.py", line 1704, in extract
    return self._extract_member(member, path, pwd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\Python\Lib\zipfile\__init__.py", line 1778, in _extract_member
    open(targetpath, "wb") as target:
    ^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'E:\\Program Files\\Programming\\platformio\\.cache\\tmp\\pkg-installing-ivafo77u\\esp32-arduino-libs\\esp32\\include\\espressif__esp_matter\\connectedhomeip\\connectedhomeip\\src\\app\\clusters\\valve-configuration-and-control-server\\valve-configuration-and-control-delegate.h'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\click\core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\click\core.py", line 1082, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\click\core.py", line 1697, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\click\core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\click\core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\run\cli.py", line 145, in cli
    process_env(
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\run\cli.py", line 208, in process_env
    ).process()
      ^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\run\processor.py", line 81, in process
    install_project_env_dependencies(
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 132, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\commands\install.py", line 149, in _install_project_env_platform
    PlatformPackageManager().install(
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\manager\platform.py", line 62, in install
    p.install_required_packages(force=force)
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\platform\_packages.py", line 76, in install_required_packages
    self.install_package(name, force=force)
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\platform\_packages.py", line 70, in install_package
    return self.pm.install(spec or self.get_package_spec(name), force=force)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\manager\_install.py", line 47, in install
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_uri(spec.uri, spec)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\manager\_install.py", line 191, in install_from_uri
    self.unpack(dl_path, tmp_dir)
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\manager\_install.py", line 42, in unpack
    return fu.unpack(dst, with_progress=False)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\unpack.py", line 179, in unpack
    self._archiver.extract_item(item, dest_dir)
  File "E:\Program Files\Programming\platformio\penv\Lib\site-packages\platformio\package\unpack.py", line 49, in extract_item
    self._afo.extract(item, dest_dir)
  File "E:\Program Files\Programming\Python\Lib\zipfile\__init__.py", line 1704, in extract
    return self._extract_member(member, path, pwd)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Program Files\Programming\Python\Lib\zipfile\__init__.py", line 1778, in _extract_member
    open(targetpath, "wb") as target:
    ^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'E:\\Program Files\\Programming\\platformio\\.cache\\tmp\\pkg-installing-ivafo77u\\esp32-arduino-libs\\esp32\\include\\espressif__esp_matter\\connectedhomeip\\connectedhomeip\\src\\app\\clusters\\valve-configuration-and-control-server\\valve-configuration-and-control-delegate.h'

```

It looks like the pkg-installing-ivafo77u directory is not written, but no error is generated to indicate that and the code is in place.

My platform.io file has the following:

```
[env:esp-wrover-kit]
;platform = espressif32@6.9.0
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = esp-wrover-kit
framework = arduino
```

I have no similar problem if I used the other commented out platform parameter entry.

Any suggestions?

Jason2866, 2025-01-25:
Platformio does not allow Whitespaces anywhere in Path from stuff it is using
```
File "E:\Program Files\Programming\....
```

lonerzzz, 2025-01-25:
Had Platformio in place for several years and was working fine, but this appeared to work. Thank you!

closed 2025-01-25 by lonerzzz