pioarduino/platform-espressif32 #111: Compile failure

lonerzzz, 2025-01-26:
I am hitting the following compile error when specifically building for esp32-c3.

```
riscv32-esp-elf-g++: fatal error: cannot execute 'E:/Prog_Files/Prog/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\esp32-c3-devkitm-1\src\Log.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'E:/Prog_Files/Prog/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\esp32-c3-devkitm-1\src\Configuration.cpp.o] Error 1
riscv32-esp-elf-g++: fatal error: cannot execute 'E:/Prog_Files/Prog/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
riscv32-esp-elf-g++: fatal error: cannot execute 'E:/Prog_Files/Prog/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory
compilation terminated.
```

My platform.io looks as follows:
```
[env:esp32-c3-devkitm-1]
;platform = espressif32@6.9.0
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = esp32-c3-devkitm-1
```

For tasmota, the discussion indicated that a fix was implemented because of path length issue on windows so I am wondering if the same issue/work around is needed here.
[Compile of Tasmota 14.x.x fails for ESP32-Cx](https://github.com/arendst/Tasmota/discussions/21964)

Jason2866, 2025-01-26:
The include shorten path length by using a prefix is not in any release version. The fix for Windows was done after the last release. It is in branch develop. Can be used with

```ini
[env:development]
platform = https://github.com/pioarduino/platform-espressif32.git#develop
board = ...
...
```

rpannekoek, 2025-02-02:
Great to hear that there will be (or kind of is) a fix for this issue on this (great) PlatformIO fork too!
However, for visibility, I would keep the issue open until the fix went into an "official" release.
I encounter this same issue with release 53.03.11 (specifically for C3), whereas it didn't happen with 51.03.07.
I've spent quite some time trying in-between releases (53.03.10 has same problem), going through discussions on Tasmota, trying several of the suggested work-arounds and figuring out what goes wrong with invoking `as.exe`, exactly (the problem is not that `as.exe` doesn't exist or that the path to `as.exe` is too long).
When I wanted to create an issue on this repo with my findings and double-checked whether such an issue wasn't already created in the meantime,  I discovered this closed issue.

Jason2866, 2025-02-02:
reopen until fix is in release as suggested from @rpannekoek

closed 2025-02-13 by Jason2866