pioarduino/platform-espressif32 #112: error compiling - no such file or directory Network.h

alessandro-satanassi, 2025-01-28:
I installed the latest version of pioarduino-platform-espressif32 .. everything installs normally and works, but when I enable wifi, during compilation it tells me that it does not find the file include network.h, but this file is there ... what could be the problem?

In file included from C:/Users/aless/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:30,
                 from C:/Users/aless/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:34,
                 from C:/Users/aless/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/AP.cpp:7:
C:/Users/aless/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:44:10: fatal error: Network.h: No such file or directory

*****************************************************************
* Looking for Network.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Network.h"
* Web  > https://registry.platformio.org/search?q=header:Network.h
*
*****************************************************************

   44 | #include "Network.h"
      |          ^~~~~~~~~~~
compilation terminated.

Jason2866, 2025-01-28:
Please provide the platformio.ini and a small example sketch where the error occurs.

alessandro-satanassi, 2025-01-28:
platformio.ini

`

[platformio]

;default_envs = rpipicow
default_envs = LILYGO_T_DONGLE_S3 

[env]

[rp2040]
platform_packages = 
	framework-arduinopico @ https://github.com/earlephilhower/arduino-pico.git#4.4.2 ;#master

[esp32]
platform_packages = 
	;platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
	;platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-lib-builder.git#master
	

[common]
lib_deps = 
	
build_unflags = 
	-std=gnu++11
	-std=gnu11

build_flags = 
	-std=gnu++17
	-std=gnu17

[display_LILYGO_T_DONGLE_S3]
build_unflags = 

build_flags = 
    -D USER_SETUP_LOADED
    -D ST7735_DRIVER
    -D SPI_FREQUENCY=50000000
    -D TFT_MISO=-1
    -D TFT_MOSI=3
    -D TFT_SCLK=5
    -D TFT_CS=4 
    -D TFT_DC=2 
    -D TFT_RST=1 
    -D TFT_WIDTH=80
    -D TFT_HEIGHT=160
    -D LOAD_GLCD  
    -D LOAD_FONT2 
    -D LOAD_FONT4 
    -D LOAD_FONT6 
    -D LOAD_FONT7 
    -D LOAD_FONT8 
    -D LOAD_GFXFF 
    -D SMOOTH_FONT
    -D ST7735_GREENTAB160x80
    ;-D TFT_RGB_ORDER=TFT_BGR
	-D TFT_RGB_ORDER=0

[display_rpipicow]
build_unflags = 

build_flags = 
    -D USER_SETUP_LOADED
    -D ST7735_DRIVER
    -D SPI_FREQUENCY=50000000
    -D TFT_MISO=-1
    -D TFT_MOSI=19
    -D TFT_SCLK=18
    -D TFT_CS=17 
    -D TFT_DC=16 
    -D TFT_RST=20 
    -D TFT_WIDTH=80
    -D TFT_HEIGHT=160
    -D LOAD_GLCD  
    -D LOAD_FONT2 
    -D LOAD_FONT4 
    -D LOAD_FONT6 
    -D LOAD_FONT7 
    -D LOAD_FONT8 
    -D LOAD_GFXFF 
    -D SMOOTH_FONT
    -D ST7735_GREENTAB160x80
    ;-D TFT_RGB_ORDER=TFT_BGR
	-D TFT_RGB_ORDER=0

[env:LILYGO_T_DONGLE_S3]
;platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
platform = https://github.com/pioarduino/platform-espressif32.git#develop
board = esp32-s3-devkitc-1
framework = arduino
board_upload.use_1200bps_touch = yes
board_upload.wait_for_upload_port = yes
board_upload.require_upload_port = yes
lib_deps = 
	adafruit/Adafruit ST7735 and ST7789 Library @ 1.10.4
	;bodmer/TFT_eSPI @ 2.5.43
	nRF24/RF24 @ 1.4.11
	;${platformio.lib_dir}/Bluetooth_OpenFIRE/ESP32
	${common.lib_deps}
build_unflags = 
	-D ARDUINO_USB_MODE=1
	-D ARDUINO_USB_CDC_ON_BOOT=1
	${common.build_unflags}
	${display_LILYGO_T_DONGLE_S3.build_unflags}
build_flags = 
	-D ARDUINO_USB_MODE=0
    -D ARDUINO_USB_CDC_ON_BOOT=0
	-D DEVICE_LILYGO_T_DONGLE_S3
	-D USES_DISPLAY
	${common.build_flags}
	${display_LILYGO_T_DONGLE_S3.build_flags}
board_build.partitions = huge_app.csv
platform_packages = 
	${esp32.platform_packages}
lib_ldf_mode = chain+
lib_compat_mode = strict

[env:rpipicow]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#develop
framework = arduino
board = rpipicow
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
monitor_speed = 9600
monitor_port = COM3 ;COM135
upload_protocol = picotool
build_type = release
lib_deps = 
	adafruit/Adafruit ST7735 and ST7789 Library @ 1.10.4
	;bodmer/TFT_eSPI @ 2.5.43
	;nRF24/RF24 @ 1.4.11
	https://github.com/nRF24/RF24.git @ 1.4.11
	${common.lib_deps}
build_unflags = 
	-D ARDUINO_USB_MODE=1
	-D ARDUINO_USB_CDC_ON_BOOT=1
	${common.build_unflags}
	${display_rpipicow.build_unflags}
build_flags = 
	-D ARDUINO_USB_MODE=0
    -D ARDUINO_USB_CDC_ON_BOOT=0
	-D USE_TINYUSB
	-D PIO_FRAMEWORK_ARDUINO_ENABLE_BLUETOOTH
	-D DUAL_CORE
	${common.build_flags}
	${display_rpipicow.build_flags}
platform_packages = ${rp2040.platform_packages}
lib_ldf_mode = chain+
lib_compat_mode = strict
`
main.cpp

`
#include <Arduino.h>
#include <SPI.h>
#include "TinyUSB_Devices.h"

#include <USB.h>

#include "pin_config_DONGLE.h"
#if defined(ARDUINO_ARCH_ESP32)
  #include <esp_now.h>
  #include <Network.h>
  #include <WiFi.h> // serve per leggere max addres
  #include <esp_mac.h>  // For the MAC2STR and MACSTR macros
#elif defined(ARDUINO_ARCH_RP2040)
  // vediamo
#endif

......................

void setup() {

...........................

WiFi.mode(WIFI_MODE_STA);
    WiFi.begin();

// Inizializza ESP-NOW
    if (esp_now_init() != ESP_OK) {
        //Serial.println("Error initializing ESP-NOW");
        tft.fillScreen(BLACK);
        tft.setTextSize(2);
        tft.setCursor(0, 0);
        tft.setTextColor(RED);
        tft.println("Error initializing ESP-NOW");
        delay (1000);
    }
    else {
        //Serial.println("ESP-NOW Inizializzato");
        tft.fillScreen(BLACK);
        tft.setTextSize(2);
        tft.setCursor(0, 0);
        tft.setTextColor(RED);
        tft.println("ESP-NOW Inizializzato");
        delay (1000);
    }

   // Imposta l'indirizzo del peer
  esp_now_peer_info_t peerInfo;
  memcpy(peerInfo.peer_addr, peerAddress, 6);
  peerInfo.channel = 0;
  peerInfo.encrypt = false;

// Aggiungi il peer
  if (esp_now_add_peer(&peerInfo) != ESP_OK) {
    //Serial.println("Failed to add peer");
    tft.setTextSize(2);
    tft.setCursor(0, 35);
    tft.setTextColor(RED);
    tft.println("Failed to add peer");
    delay (1000);
  }
  else {
Serial.println("add peer OK");
    tft.setTextSize(2);
    tft.setCursor(0, 35);
    tft.setTextColor(RED);
    tft.println("add peer OK");
    delay (1000);
  }

............................
}

`

alessandro-satanassi, 2025-01-28:
I get the same error on both win11 and linux fedora 41, with the latest version of vscode with platformio extension

Jason2866, 2025-01-28:
If this setting `lib_ldf_mode = chain+` is done Platformio goes crazy. Never say this working as it should and described in docu. Change to `lib_ldf_mode = chain` If this does not work one of your libs does not correctly declare dependencies in its `library.json`

Not a pioarduino issue. The problem is the LDF (Library Dependency Finder) from Platformio Core.

Jason2866, 2025-01-28:
Btw. your platformio.ini has in general entrys not needed. For example

```
board_upload.use_1200bps_touch = yes
board_upload.wait_for_upload_port = yes
board_upload.require_upload_port = yes
```
That's just garbage, since TinyUSB is not used where this is maybe needed.
Clean up your platformio.ini!

Jason2866, 2025-01-28:
Closing since the issues are caused from wrong setup.

closed 2025-01-28 by Jason2866

alessandro-satanassi, 2025-01-28:
Thanks, with the changes you indicated, it works now.
Very kind... thanks for the excellent work you carry out and compliments for your expertise