pioarduino/platform-espressif32 #113: ESP32-H2 - how to define some parameters?

dattasaurabh82, 2025-02-06:
First of all - Thank you so much now that we can use the new boards like ESP32-H2
So, for ESP32-H2, the board settings are defined in in the boards/esp32-h2-devkitm-1.json is and that is: 

```json
{
  "build": {
    "core": "esp32",
    "f_cpu": "96000000L",
    "f_flash": "64000000L",
    "f_image": "48000000L",
    "flash_mode": "qio",
    "mcu": "esp32h2",
    "variant": "esp32h2",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ]
  },
  "connectivity": [
    "bluetooth",
    "zigbee"
  ],
  "debug": {
    "openocd_target": "esp32h2.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-H2-DevKit",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32h2/esp32-h2-devkitm-1/index.html",
  "vendor": "Espressif"
}
```
How can I add these as seen in Arduino IDE settings, but in the pio ini file?
Arduino User Space reversible Fuse settings:

```txt
Board: ESP32-H2-Dev Module
Upload Speed: 921600
CDC On Boot: Disabled
Flash Frequency: 64MHz
Flash Mode: QIO
Flash Size: 4MB (32Mb)
Partition Scheme: Default 4MB (1.2MB APP/1.5MB SPIFFS)
Erase Flash: Disabled
JTAG Adapter: Disabled
Zigbee Mode: Disabled
```

My current ini is:

```ini
[env:esp32-h2-devkitm-1]
platform = https://github.com/pioarduino/platform-espressif32.git
board = esp32-h2-devkitm-1
framework = arduino
lib_deps = 
    Adafruit NeoPixel

monitor_speed = 115200  ; Monitor Settings
upload_speed = 921600  ; Upload Settings

board_build.f_flash = 64000000L  ; Flash Frequency
board_build.flash_mode = qio
board_build.flash_size = 4MB

board_build.partitions = partitions/minimal.csv  ; Partition Settings
board_build.cdc_on_boot = no  ; CDC Settings

; Flash Erase Settings
; upload_flags = 
;     --erase-all=no

; JTAG and Zigbee Settings
; Clear configs before redefining
build_unflags =
    -DCONFIG_ESP_DEBUG_OCDAWARE
    -DCONFIG_ZB_ENABLED
    -DCONFIG_FREERTOS_ASSERT_DISABLE
    -DCONFIG_TINYUSB_DEBUG_LEVEL

; JTAG and Zigbee Settings
build_flags =
    -DCONFIG_ZB_ENABLED=0
    -DCONFIG_TINYUSB_DEBUG_LEVEL=0
    -DCONFIG_ESP_DEBUG_OCDAWARE=0
```

Jason2866, 2025-02-07:
The most important is missing. What do you want to compile? For a Blink sketch nothing needs to be added. If you want to use zigbee see example https://github.com/pioarduino/platform-espressif32/blob/main/examples/arduino-zigbee-light/platformio.ini

dattasaurabh82, 2025-02-07:
> The most important is missing. What do you want to compile? For a Blink sketch nothing needs to be added. If you want to use zigbee see example https://github.com/pioarduino/platform-espressif32/blob/main/examples/arduino-zigbee-light/platformio.ini

I'm writing a piece of code, that needs to run in low-power profile, so want to optimize as many things as possible. 
I have tested the code base from Arduino IDE (with the aforementioned fuse settings) and now want to port the code base to PIO with the same fuse settings.  

So looking at the above example @Jason2866 , few things popped out, please confirm:

1. `board_build.partitions` are not available as variables and the respective `partitions.csv` file need to be placed and pointed to in the ini file?  
2. `board_build.filesystem`: what is that and so partitions are separated from file system unlike how Arduino IDE does it? 
3. How can we do **Erase Flash before upload**?  In Arduino IDE that is: `Erase Flash: ENABLED/DISBALED` 

| --- | ---| 
| Arduino IDE Fuse Setting | PIO ini settings |  
| --- | --- |
| Board: ESP32-H2-Dev Module  | `board = esp32-h2-devkitm-1` |
| Upload Speed: 921600  | `upload_speed = 921600` |
| CDC On Boot: Disabled  | `board_build.cdc_on_boot = no` |
| Flash Frequency: 64MHz  | `board_build.f_flash = 64000000L` |
| Flash Mode: QIO  | `board_build.flash_mode = qio` |
| Flash Size: 4MB (32Mb)  | `board_build.flash_size = 4MB` |
| JTAG Adapter: Disabled | `build_flags = ` |
|  | `-DCONFIG_TINYUSB_DEBUG_LEVEL=0` |
|  | `-DCONFIG_ESP_DEBUG_OCDAWARE=0` |
| Zigbee Mode: Disabled | `build_flags = ` |
|  | `-DCONFIG_ZB_ENABLED=0` or  |
|  | or no need to put it here if  `-DZIGBEE_MODE_ED ` enables zigbee ? |
|  Partition Scheme: Default 4MB (1.2MB APP/1.5MB SPIFFS) | partitions file copied next to the project dir: `board_build.partitions=partitions.csv` ?  |  
| Erase Flash: Disabled | ?  |

sivar2311, 2025-02-07:
> 1 `board_build.partitions` are not available as variables and the respective partitions.csv file need to be placed and pointed to in the ini file?

They are available as files and are part of the platform - See [available partition layouts](https://github.com/espressif/arduino-esp32/tree/master/tools/partitions)

You can simply set one of the available filenames in platformio.ini:
`board_build.partitions = filename.csv`

Please read [PlatformIO Documentation -  Partition Tables](https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables)

> 2. `board_build.filesystem`: what is that and so partitions are separated from file system unlike how Arduino IDE does it?

Please read [PlatformIO Documentation - Uploading files to file system](https://docs.platformio.org/en/latest/platforms/espressif32.html#uploading-files-to-file-system)

> 3. How can we do Erase Flash before upload? In Arduino IDE that is: Erase Flash: ENABLED/DISBALED

Click the PIO - Icon / PROJECT TASKS / <name of your environment> / Platform / Erase Flash:

[image]

For your big table:
Only change things which you really need to change and if you know what you're doing.

If you need to change a setting please ask for a specific setting you want to change **and** tell the reason **why**  you want to change.

Jason2866, 2025-02-07:
Closing since github issues is not for general questions. Please visit our [Discord](https://discord.gg/Nutz9crnZr) server for general help.

closed 2025-02-07 by Jason2866