pioarduino/platform-espressif32 #116: Easy way to switch between sdkconfigs

Bascy, 2025-02-14:
By adding espidf to the framework property in platformio.ini we can start the menuconfig and change properties of the underlying framework, i.e. enabling task tracing and cpu use reports.

When i do this i noticed a sdkconfig files is created in the root of my project, but the sdkconfig in the actual framework folder is updated.
What i would like is a way to switch from one sdkconfig to another (like enable task tracing in debug but not in release), i.e. by a property in the platformio, or a pre-build script, but i don't know how to do this.

Can anyone help?

Jason2866, 2025-02-14:
This is not the case
```
When i do this i noticed a sdkconfig files is created in the root of my project, but the sdkconfig in the actual framework folder is updated.
```

What you want to achieve is not clearly described / unclear and mixed with false assumptions.
Without specifying exactly whats your goal, can't help.
Describe the goal and not what you think it is needed in setup. That's the second step.

Bascy, 2025-02-14:
Can you kindly direct me to the correct Discord, i do not see any references to it here

Jason2866, 2025-02-14:
"Readme" click the Discord badge. https://discord.gg/Nutz9crnZr

closed 2025-02-17 by Jason2866