pioarduino/platform-espressif32 #118: Stable release install v3.1.2 instead of v3.1.3

rev1204, 2025-02-21:
I'm copy pasted the platform from README that currently espressif Arduino 3.1.3:

`platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip`

But the version installed is v3.1.2

Jason2866, 2025-02-21:
Close VSC. Delete hidden folder `.platformio/platforms` Start VSC and it will install 3.1.3

closed 2025-02-21 by Jason2866

jnovonj, 2025-02-24:
Hi,

  I apologize to reopen this.

Same problem here.

```
[platformio]
name = ESP32-S3 Pi Zero 2 W
description = Project ESP32-S3 Pi Zero 2 W
default_envs = main

[env:main]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = ESP32-S3 Pi Zero 2 W
framework = arduino
```

1.- Close Visual Studio Code 1.97.2
2.- Delete %USERPROFILE%\.platformio\platforms
3.- Rerun Visual Studio Code 1.97.2

```
Resolving main dependencies...
Platform Manager: Installing https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Platform Manager: **espressif32@53.3.11** has been installed!
UnknownBoard: Unknown board ID 'ESP32-S3 Pi Zero 2 W'             <== _I must copy this file, this is known._
```

After copy my file I can compile with the versión:

```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.1.1 
 - framework-arduinoespressif32-libs @ 5.3.0+sha.cfea4f7c98 
 - tool-esptoolpy @ 4.8.5 
 - tool-mklittlefs @ 3.2.0 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530

```

The file %USERPROFILE%\.platformio\platforms\espressif32\README.md has a SHA1256:

5688668b9ee4802545310e9aed955442b0a8a7412389f7480b796147faaae2ef

Best Regards.

jnovonj, 2025-02-24:
I come back with the solution for my problem:

1.- Close Visual Studio Code 1.97.2
2.- Delete %USERPROFILE%\.platformio
3.- Rerun Visual Studio Code 1.97.2

```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.1.3 
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
```

Jason2866, 2025-02-24:
Delete additionally folder ` %USERPROFILE%.platformio\.cache`