pioarduino/platform-espressif32 #120: I2S DMA error

schreibfaul1, 2025-02-23:
Since Arduino version 3.1.2, no memory is provided for I2S:

Example:
````c++
#include "Arduino.h"
#include "esp_err.h"
#include "driver/i2s_std.h"

void setup(){
    esp_err_t err;
    i2s_chan_handle_t i2s_tx_handle;
    log_w("Arduino Version: %d.%d.%d", ESP_ARDUINO_VERSION_MAJOR, ESP_ARDUINO_VERSION_MINOR, ESP_ARDUINO_VERSION_PATCH);
    log_w("ESP-IDF Version: %d.%d.%d", ESP_IDF_VERSION_MAJOR, ESP_IDF_VERSION_MINOR, ESP_IDF_VERSION_PATCH);

    i2s_chan_config_t chan_config;
    chan_config.id            = I2S_NUM_AUTO;           // I2S_NUM_AUTO, I2S_NUM_0, I2S_NUM_1
    chan_config.role          = I2S_ROLE_MASTER;        // I2S controller master role, bclk and lrc signal will be set to output
    chan_config.dma_desc_num  = 32;                     // number of DMA buffer
    chan_config.dma_frame_num = 256;                    // I2S frame number in one DMA buffer.
    chan_config.auto_clear    = true;                   // i2s will always send zero automatically if no data to send
    err = i2s_new_channel(&chan_config, &i2s_tx_handle, NULL);
    if(err != ESP_OK) log_e("i2s_new_channel failed: %s", esp_err_to_name(err));

    i2s_std_config_t std_config;
    std_config.slot_cfg                = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO); // Set to enable bit shift in Philips mode
    std_config.gpio_cfg.bclk           = I2S_GPIO_UNUSED;           // BCLK, Assignment in setPinout()
    std_config.gpio_cfg.din            = I2S_GPIO_UNUSED;           // not used
    std_config.gpio_cfg.dout           = I2S_GPIO_UNUSED;           // DOUT, Assignment in setPinout()
    std_config.gpio_cfg.mclk           = I2S_GPIO_UNUSED;           // MCLK, Assignment in setPinout()
    std_config.gpio_cfg.ws             = I2S_GPIO_UNUSED;           // LRC,  Assignment in setPinout()
    std_config.clk_cfg.sample_rate_hz  = 44100;
    std_config.clk_cfg.clk_src         = I2S_CLK_SRC_DEFAULT;        // Select PLL_F160M as the default source clock
    std_config.clk_cfg.mclk_multiple   = I2S_MCLK_MULTIPLE_128;      // mclk = sample_rate * 256
    err = i2s_channel_init_std_mode(i2s_tx_handle, &std_config);
    if(err != ESP_OK) log_e("i2s_channel_init_std_mode failed: %s", esp_err_to_name(err));
}

void loop(){
    vTaskDelay(1000 / portTICK_PERIOD_MS);
}
````
For the test an **ESP32-S3** board without external circuitry is used (no SD, no DAC etc.) 
it works up to and including Arduino V3.1.1
````
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip ; Arduino 3.1.1 / IDF 5.3.2
framework = arduino
````
it does not work with Arduino V3.1.2. or 3.1.3
```` 
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.12/platform-espressif32.zip ; Arduino 3.1.2 / IDF 5.3.2 
;platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip ; Arduino 3.1.3 / IDF 5.3.2
````
the error message:

[image]

One more note: Arduino version 3.1.2 and 3.1.3 run perfectly in the Arduino IDE. And it apparently only affects the ESP32-S3

sivar2311, 2025-02-23:
@schreibfaul1 The bug seems to be related to PSRAM
See https://github.com/schreibfaul1/ESP32-audioI2S/issues/975#issuecomment-2676733074
and https://github.com/schreibfaul1/ESP32-audioI2S/issues/975#issuecomment-2676740885

schreibfaul1, 2025-02-23:
@sivar2311,
you're right, I had experimented with an ESP-S3-12k kit and didn't realise the connection with the PSRAM. I also have the error with an ESP-S3 N16R8.
And therefore the cause does not lie with PioArduino.

closed 2025-02-23 by schreibfaul1