pioarduino/platform-espressif32 #121: Using latest core 3.x but using old 2.x boards?

UnexpectedMaker, 2025-03-08:
Hi folks - thanks for you awesome work on supporting `Arduino ESP32 Core 3.x` on PlatformIO.

I've had a customer contact me letting me know that boards I've had added to the `Arduino ESP32 Core 3.x` are not available in PIO using `pioarduino` and it indeed seems that the boards list here is the Core 2.x boards, not core 3.x ones.

How are users supposed to use newer board definitions added to Core 3.x for Arduino? Is there an install step we are missing?

Thanks :)

Xylopyrographer, 2025-03-09:
You'll need to open a PR to add them. See https://github.com/pioarduino/platform-espressif32/pull/86

UnexpectedMaker, 2025-03-09:
Ok, thanks, I'll get them added and raise a PR.

Jason2866, 2025-03-09:
PR has to be done against branch develop. See github README.md

UnexpectedMaker, 2025-03-10:
Thanks for the heads up  @Jason2866 

PR raised, cheers :)

closed 2025-03-10 by UnexpectedMaker