pioarduino/platform-espressif32 #124: Not able to use pioarduino platform

kumardeo, 2025-03-21:
Hey, thanks for this project.
I am new to IoT and C++ world (coming from JavaScript).
I need to use `Arduino Core v3.*`. When I use the following `platformio.ini`, I get various errors reported by VS Code as well as the code does not compile.

```ini
[env:arduino_nano_esp32]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = arduino_nano_esp32
framework = arduino
```
I thought I need to use `pioarduino IDE` extension instead of official `PlatformIO IDE` in VS Code. Therefore, I uninstalled the official `PlatformIO IDE`, deleted `.platformio` folder, installed the `pioarduino IDE` and let extension setup everything. But, it did not work.
VS Code Intellisense shows the following errors:

[image]

```
expression preceding parentheses of apparent call must have (pointer-to-) function typeC/C++(109)

#define pinMode(pin,mode) pinMode(digitalPinToGPIONumber(pin), mode)
Expands to:
pinMode(digitalPinToGPIONumber(LED_BUILTIN), 0x03)
```

```
expression preceding parentheses of apparent call must have (pointer-to-) function typeC/C++(109)

#define digitalWrite(pin,val) digitalWrite(digitalPinToGPIONumber(pin), val)
Expands to:
digitalWrite(digitalPinToGPIONumber(LED_BUILTIN), 0x0)
```

When I try to compile using the build button, it logs (I have omitted many to keep it short):

```
# more logs here but I omitted it

/home/deo/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:251:21: error: expected primary-expression before '_pin'
  251 | void noTone(uint8_t _pin);
      |                     ^~~~
/home/deo/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:20:65: note: in definition of macro 'noTone'
   20 | #define noTone(_pin)              noTone(digitalPinToGPIONumber(_pin))
      |                                                                 ^~~~
src/main.cpp: In function 'void setup()':
/home/deo/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:47:49: error: 'pinMode' was not declared in this scope
   47 | #define pinMode(pin, mode)                      pinMode(digitalPinToGPIONumber(pin), mode)
      |                                                 ^~~~~~~
src/main.cpp:6:3: note: in expansion of macro 'pinMode'
    6 |   pinMode(LED_BUILTIN, OUTPUT);
      |   ^~~~~~~
/home/deo/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:47:49: note: maximum limit of 1000 namespaces searched for 'pinMode'
   47 | #define pinMode(pin, mode)                      pinMode(digitalPinToGPIONumber(pin), mode)
      |                                                 ^~~~~~~
src/main.cpp:6:3: note: in expansion of macro 'pinMode'
    6 |   pinMode(LED_BUILTIN, OUTPUT);
      |   ^~~~~~~
src/main.cpp: In function 'void loop()':
/home/deo/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:46:49: error: 'digitalWrite' was not declared in this scope
   46 | #define digitalWrite(pin, val)                  digitalWrite(digitalPinToGPIONumber(pin), val)
      |                                                 ^~~~~~~~~~~~
src/main.cpp:11:3: note: in expansion of macro 'digitalWrite'
   11 |   digitalWrite(LED_BUILTIN, LOW);
      |   ^~~~~~~~~~~~
/home/deo/.platformio/packages/framework-arduinoespressif32/cores/esp32/io_pin_remap.h:46:49: note: maximum limit of 1000 namespaces searched for 'digitalWrite'
   46 | #define digitalWrite(pin, val)                  digitalWrite(digitalPinToGPIONumber(pin), val)
      |                                                 ^~~~~~~~~~~~
src/main.cpp:11:3: note: in expansion of macro 'digitalWrite'
   11 |   digitalWrite(LED_BUILTIN, LOW);
      |   ^~~~~~~~~~~~
*** [.pio/build/arduino_nano_esp32/src/main.cpp.o] Error 1
*** [.pio/build/arduino_nano_esp32/lib3c7/Matter/Matter.cpp.o] Error 1
===================================================== [FAILED] Took 5.54 seconds =====================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

It looks like I am not setting up things correctly, what is the correct way to using pioarduino?
Also, do I only need to use `pioarduino IDE` extension or it is optional?
Do I only need to change `platform` field in `platformio.ini`?
What are the requirements?

sivar2311, 2025-03-21:
To get Arduino 3.x all you need to do is to change the platform setting in platformio.ini.

pioarduinoIDE is optional.

Jason2866, 2025-03-21:
Idk if pioarduino is working with the board `arduino_nano_esp32`. The board is weird since it is different to all other espressif boards. Arduino people decided not to use the uploader `esptool.py`. The programmed there own uploader utility which is old, outdated and unreliable.

Did the implementation of the Arduino nano tool `dfuutil` without the possibility to test, since i don't have an Arduino nano board and i will not buy one (overpriced).

kumardeo, 2025-03-21:
@Jason2866 You are right, I tried selecting another board and it worked, but I have an Arduino Nano ESP32.
Thanks for your reply, looks like I can't use PlatformIO IDE (I don't really like Arduino IDE).

Jason2866, 2025-03-21:
I think you can use the board with pioarduino, just don't use the `arduino_nano_esp32` as boards manifest. Use `esp32dev` as board and just flash it with pioarduino. It will just work.

Jason2866, 2025-03-21:
@kumardeo Can you provide an easy example code which fails? I can't promise if i could fix (looks like an Arduino core issue). But i will have a look.

TD-er, 2025-03-21:
Is `LED_BUILTIN` defined here?

Jason2866, 2025-03-21:
This setting has to be changed to `BOARD_USES_HW_GPIO_NUMBERS` https://github.com/espressif/arduino-esp32/blob/6c04a9315309b530120de98321f8d900c151a9b9/variants/arduino_nano_nora/pins_arduino.h#L15
either by changing the boards manifest or in platformio.ini by overwriting the boards manifest entry

Jason2866, 2025-03-21:
I will change the Nano boards manifest in branch development (change will be in release core Arduino 3.2.0) to do no remapping and use `BOARD_USES_HW_GPIO_NUMBERS` so the example sketches from espressif Arduino core repo will (hopefully) compile.

Jason2866, 2025-03-21:
@kumardeo 

Try:

```
[env:arduino_nano_esp32]
platform = https://github.com/pioarduino/platform-espressif32.git#develop
board = arduino_nano_esp32
framework = arduino
```

have changed in actual develop branch to use HW GPIO numbers for the Nano board. Maybe you have to `#define LED_BUILTIN`

Jason2866, 2025-03-21:
Fixed in branch development

closed 2025-03-21 by Jason2866

Jason2866, 2025-03-21:
Yes, use espressif hardware gpio numbers, so 17 is correct

kumardeo, 2025-03-21:
Thanks again!