pioarduino/platform-espressif32 #125: WiFi Provisioning with BLE throws runtime error when using PlatformIO

kumardeo, 2025-03-22:
I am not sure if this is the correct repo to open this issue.

> If it is irrelevant, feel free to close this issue.

When using `PlatformIO IDE (platform: pioarduino | core: v3.1.3)` and `ESP32 Dev Module (nodemcu-32s)` board.
> [!NOTE]
> The same code works as expected when using `Arduino IDE (core: v3.1.3)`.

The runtime error:

```
E (247) simple_ble: simple_ble_start enable controller failed 259
E (248) protocomm_ble: simple_ble_start failed w/ error code 0x103
E (249) network_prov_scheme_ble: Failed to start protocomm BLE service
E (255) network_prov_mgr: Failed to start service
[   265][E][WiFiProv.cpp:165] beginProvision(): network_prov_mgr_start_provisioning failed!
```

`src/main.cpp`
```cpp
#include <Arduino.h>
#include <WiFiProv.h>

const uint8_t LED_PIN = 2;
const char *pop = "12345678";                // Proof of possession - otherwise called a PIN - string provided by the device, entered by the user in the phone app
const char *service_name = "PROV_MYAPP";  // Name of your device (the Espressif apps expects by default device name starting with "Prov_")
const char *service_key = NULL;              // Password used for SofAP method (NULL = no password needed)
bool reset_provisioned = true;               // When true the library will automatically delete previously provisioned data.

void setup() {
  pinMode(2, OUTPUT);

  Serial.begin(115200);

  Serial.println("Begin Provisioning using BLE");
  // Sample uuid that user can pass during provisioning using BLE
  uint8_t uuid[16] = { 0xb4, 0xdf, 0x5a, 0x1c, 0x3f, 0x6b, 0xf4, 0xbf, 0xea, 0x4a, 0x82, 0x03, 0x04, 0x90, 0x1a, 0x02 };
  WiFiProv.beginProvision(
    NETWORK_PROV_SCHEME_BLE, NETWORK_PROV_SCHEME_HANDLER_FREE_BLE, NETWORK_PROV_SECURITY_1, pop, service_name, service_key, uuid, reset_provisioned);
  Serial.println("You may use this BLE QRCode:");
  WiFiProv.printQR(service_name, pop, "ble");
}

void loop() {
  bool isConnected = WiFi.isConnected();
  if (isConnected) {
    digitalWrite(LED_PIN, HIGH);
  } else {
    digitalWrite(LED_PIN, LOW);
    delay(200);
    digitalWrite(LED_PIN, HIGH);
    delay(200);
  }
}

```

`platformio.ini`
```ini
[env:esp32dev]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip
board = esp32dev
framework = arduino
board_build.partitions = rainmaker_4MB_no_ota.csv
monitor_speed = 115200
```

Jason2866, 2025-03-22:
Highly unlikely that it works with ArduinoIDE and not with pioarduino. The source code and the static libs are 100% same. Probably different core or settings used.

kumardeo, 2025-03-22:
I am using same core `v3.1.3` (PlatformIO IDE: `pioarduino v53.03.13 (core v3.1.3)` | Arduino IDE: `core v3.1.3`).
For settings, I only changed partition scheme in both the IDEs to `Rainmaker 4MB No OTA`.

Jason2866, 2025-03-22:
Settings done in ArduinoIDE? Board? Best post a screenshot of

Jason2866, 2025-03-22:
Please try this wifi provision sketch. In yours the NVS status is unknown.

```arduino 
/*
Please read README.md file in this folder, or on the web:
https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFiProv/examples/WiFiProv

Note: This sketch takes up a lot of space for the app and may not be able to flash with default setting on some chips.
  If you see Error like this: "Sketch too big"
  In Arduino IDE go to: Tools > Partition scheme > chose anything that has more than 1.4MB APP
   - for example "No OTA (2MB APP/2MB SPIFFS)"
*/

#include "sdkconfig.h"
#if CONFIG_ESP_WIFI_REMOTE_ENABLED
#error "WiFiProv is only supported in SoCs with native Wi-Fi support"
#endif

#include "WiFiProv.h"
#include "WiFi.h"

// #define USE_SOFT_AP // Uncomment if you want to enforce using the Soft AP method instead of BLE
const char *pop = "abcd1234";           // Proof of possession - otherwise called a PIN - string provided by the device, entered by the user in the phone app
const char *service_name = "PROV_123";  // Name of your device (the Espressif apps expects by default device name starting with "Prov_")
const char *service_key = NULL;         // Password used for SofAP method (NULL = no password needed)
bool reset_provisioned = true;          // When true the library will automatically delete previously provisioned data.

// WARNING: SysProvEvent is called from a separate FreeRTOS task (thread)!
void SysProvEvent(arduino_event_t *sys_event) {
  switch (sys_event->event_id) {
    case ARDUINO_EVENT_WIFI_STA_GOT_IP:
      Serial.print("\nConnected IP address : ");
      Serial.println(IPAddress(sys_event->event_info.got_ip.ip_info.ip.addr));
      break;
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED: Serial.println("\nDisconnected. Connecting to the AP again... "); break;
    case ARDUINO_EVENT_PROV_START:            Serial.println("\nProvisioning started\nGive Credentials of your access point using smartphone app"); break;
    case ARDUINO_EVENT_PROV_CRED_RECV:
    {
      Serial.println("\nReceived Wi-Fi credentials");
      Serial.print("\tSSID : ");
      Serial.println((const char *)sys_event->event_info.prov_cred_recv.ssid);
      Serial.print("\tPassword : ");
      Serial.println((char const *)sys_event->event_info.prov_cred_recv.password);
      break;
    }
    case ARDUINO_EVENT_PROV_CRED_FAIL:
    {
      Serial.println("\nProvisioning failed!\nPlease reset to factory and retry provisioning\n");
      if (sys_event->event_info.prov_fail_reason == NETWORK_PROV_WIFI_STA_AUTH_ERROR) {
        Serial.println("\nWi-Fi AP password incorrect");
      } else {
        Serial.println("\nWi-Fi AP not found....Add API \" nvs_flash_erase() \" before beginProvision()");
      }
      break;
    }
    case ARDUINO_EVENT_PROV_CRED_SUCCESS: Serial.println("\nProvisioning Successful"); break;
    case ARDUINO_EVENT_PROV_END:          Serial.println("\nProvisioning Ends"); break;
    default:                              break;
  }
}

void setup() {
  Serial.begin(115200);
  WiFi.begin();  // no SSID/PWD - get it from the Provisioning APP or from NVS (last successful connection)
  WiFi.onEvent(SysProvEvent);

// BLE Provisioning using the ESP SoftAP Prov works fine for any BLE SoC, including ESP32, ESP32S3 and ESP32C3.
#if CONFIG_BLUEDROID_ENABLED && !defined(USE_SOFT_AP)
  Serial.println("Begin Provisioning using BLE");
  // Sample uuid that user can pass during provisioning using BLE
  uint8_t uuid[16] = {0xb4, 0xdf, 0x5a, 0x1c, 0x3f, 0x6b, 0xf4, 0xbf, 0xea, 0x4a, 0x82, 0x03, 0x04, 0x90, 0x1a, 0x02};
  WiFiProv.beginProvision(
    NETWORK_PROV_SCHEME_BLE, NETWORK_PROV_SCHEME_HANDLER_FREE_BLE, NETWORK_PROV_SECURITY_1, pop, service_name, service_key, uuid, reset_provisioned
  );
  log_d("ble qr");
  WiFiProv.printQR(service_name, pop, "ble");
#else
  Serial.println("Begin Provisioning using Soft AP");
  WiFiProv.beginProvision(NETWORK_PROV_SCHEME_SOFTAP, NETWORK_PROV_SCHEME_HANDLER_NONE, NETWORK_PROV_SECURITY_1, pop, service_name, service_key);
  log_d("wifi qr");
  WiFiProv.printQR(service_name, pop, "softap");
#endif
}

void loop() {}
```

Jason2866, 2025-03-22:
Working.

[image]

Jason2866, 2025-03-22:
The messages seen are normal in the process. Probably the are not fetched in ArduinoIDE

kumardeo, 2025-03-22:
> The messages seen are normal in the process. Probably the are not fetched in ArduinoIDE

When flashed using PlatformIO, it prints QR and runs program but it does not start provisioning over BLE.
When flashed using Arduino and connect to serial monitor in PlatformIO, it does not print any errors and provisioning works as normal.

The error occurs and provisioning app cannot find device if flashed using PlatformIO. Works without error and issues when flashing through Arduino IDE.

Jason2866, 2025-03-22:
You are right. Honestly i have no idea where the bug is?! The BLE part is a static precompiled library. The same is used for ArduinoIDE and pioarduino. Since no error in build and linking process happens idk what goes wrong.

kumardeo, 2025-03-22:
I'm a beginner in C++. I was using JavaScript before, so I don’t have much experience with how C++ works, especially with the compiler and linking process.

Jason2866, 2025-03-22:
Status: Did overflash the plain firmware at address 0x10000 
The error moves or gets fixed (cross checked). So the plain build firmware is faulty.

Jason2866, 2025-03-23:
Old issue. https://github.com/espressif/arduino-esp32/issues/8176

Jason2866, 2025-03-23:
pending PR to fix https://github.com/espressif/arduino-esp32/pull/11167

kumardeo, 2025-03-23:
Just curious, I know both Arduino IDE and pioarduino uses exact same core then why do we need to comment (or remove) those lines to make it work on pioarduino while Arduino IDE does not require to remove those lines from core?

Jason2866, 2025-03-23:
The build engine from pioarduino is different. It seems the weak declaration does not works the same way in the pre process compiler

kumardeo, 2025-03-23:
I have more to learn.
Thank you! 😀

Jason2866, 2025-03-23:
Luckily not an usually bug. That's an weird one.

Jason2866, 2025-04-07:
Fix for pioarduino is to add `lib_archive = no` to the platformio.ini See example https://github.com/pioarduino/test-rainmaker/tree/wifiprov

Still it is an issue when compiling Arduino as an component of IDF with the provided espressif setup.
So closing this issus and the open issue in Arduino repo is still relevant.

closed 2025-04-07 by Jason2866

Jason2866, 2025-04-07:
Default setting for `lib_archive` changed to `False`. Change is done in branch develop.
Will be in next release.

kumardeo, 2025-04-07:
Thank you so much for taking the time to work on this issue!