pioarduino/platform-espressif32 #126: Error in I2S nearly untrackable because of no error output via USB

Kybernaut000, 2025-03-24:
Hi,

I had a really hard time tracking down this error on my Arduino Nano ESP32:
[https://github.com/espressif/arduino-esp32/issues/11004](https://github.com/espressif/arduino-esp32/issues/11004)

I'm using this board in Platform IO with pioarduino, current version (see below). I'm using `Serial.setDebugOutput(true);` to redirect the ESP_LOGx to the `Serial` ouput, which is basically the USB output (`Serial` is of type `USBCDC` on this board).

I only received a very small messsage `initPDMtx(): ERROR: ESP_ERR_INVALID_ARG` and `begin(): I2S initialization failed`, coming from `ESP_I2S.cpp`, which didn't give me a good hint.

The call stack is :
```
I2SClass::begin()
  I2SClass::initPDMtx()
    i2s_pdm.c - i2s_channel_init_pdm_tx_mode()
```

Normally `i2s_channel_init_pdm_tx_mode()` should output an error message (everything is in `ESP_RETURN_ON_FALSE`'s, which should print to stderr or stdout?), but I couldn't see this via Serial, although I was using `Serial.setDebugOutput(true);`. At the same time, as this function is part of the precompiled libs, I could not activate debugging or put in some printf's or whatever.

I didn't want to learn how to manually compile all that stuff and produce my own libs. So I copied a lot of stuff from `i2s_pdm.c` to `ESP_I2S.cpp`, and got the error message I was needing.

My assumption is, that `Serial.setDebugOutput(true);` doesn't work on precompiled libs. Am I right? Am I completely wrong and just didn't find the correct build_flag? What could have been an easier way to spot / receive this message?

(Also just for my curiosity: why are you precompiling hte libs? Error tracking would be easier, when having the source code locally and in the compile chain.)

Regards,
Alex

**Part of build messages for version info:**
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/arduino_nano_esp32.html
PLATFORM: Espressif 32 (53.3.13) > Arduino Nano ESP32
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, dfu, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.1.3
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a
```

Kybernaut000, 2025-03-24:
Hi!

I added some info.
Regards,

Alex

## Additional notes:
- In the mentioned link [https://github.com/espressif/arduino-esp32/issues/11004](https://github.com/espressif/arduino-esp32/issues/11004) they are talking about a general issue, which also comes up when using `I2S_MODE_STD`. In my case, it only comes up with `I2S_MODE_PDM_TX`.
- I made a minimal example, please find below. I also attached the output on Serial Monitor.

## Contents
You'll find below:

- Serial outputs
  - Serial output with standard ESP_I2S.cpp, with `I2S_MODE_PDM_TX` --> error, but not from i2s_pdm.c
  - Serial output with heavily modified ESP_I2s.cpp, with `I2S_MODE_PDM_TX` --> error
  - Serial output with heavily modified ESP_I2s.cpp, with `I2S_MODE_STD` --> no error
- Sources
  - Source of main.cpp
  - Source of platform.ini

## Serial outputs
### Serial output with standard ESP_I2S.cpp, with `I2S_MODE_PDM_TX`

The "10.20.30.40.50.60.70.71." comes from debug prints I added to ESP_I2S.cpp:
- The "71." is printed just before the call to
`I2S_ERROR_CHECK_RETURN_FALSE(i2s_channel_init_pdm_tx_mode(tx_chan, &i2s_pdm_tx_config))`
- the following print "72." after that line is not printed any more.

```
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 0.02
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   374060 B ( 365.3 KB)
  Free Bytes        :   329480 B ( 321.8 KB)
  Allocated Bytes   :    39196 B (  38.3 KB)
  Minimum Free Bytes:   324752 B ( 317.1 KB)
  Largest Free Block:   262132 B ( 256.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8385672 B (8189.1 KB)
  Allocated Bytes   :      576 B (   0.6 KB)
  Minimum Free Bytes:  8385640 B (8189.1 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
  Bus Mode          : OPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
[... 123 lines cut ...]
### platformio.ini
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:arduino_nano_esp32]
# platform = espressif32
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = arduino_nano_esp32
framework = arduino

build_unflags =
    -DBOARD_HAS_PIN_REMAP

build_flags = 
    -DCORE_DEBUG_LEVEL=5

```

Jason2866, 2025-03-24:
Which issue is caused by pioarduino here? pioarduino can't solve problems caused py Arduino core. pioarduino is just an other IDE. The underlying used source code is https://github.com/espressif/arduino-esp32

closed 2025-03-24 by Jason2866

Kybernaut000, 2025-03-24:
Update, trying other releases / tags:

### RC1
platform = https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20-rc1/platform-espressif32.zip
Same error
  [  2508][E][ESP_I2S.cpp:552] initPDMtx(): ERROR: ESP_ERR_INVALID_ARG
  [  2508][E][ESP_I2S.cpp:699] begin(): I2S initialization failed.
  Compile Date/Time : Mar 24 2025 11:06:42
  ESP-IDF Version   : v5.4-561-gbcb3c32d3a-dirty
  Arduino Version   : 3.2.0

### RC2
platform = https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20-rc2/platform-espressif32.zip
Works!

### Develop
platform = https://github.com/pioarduino/platform-espressif32.git#develop
Works!

### Still:
How to track those errors? I still don't get debug prints, when inputting trash:
`if (!i2s.begin(I2S_MODE_PDM_TX, 1'000'000, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO))`
doesn't give error prints from i2s_pdm.c.

Kybernaut000, 2025-03-24:
Sorry Jason, didn't see your answer, was to quick with commenting.

I'm not aware whom to blame. My problem is not the malfunctioning I2S-library, that's none of your business.

What I'm asking is: How to track those errors down? As I understand, the libraries are compiled, and there are no debug prints coming from that; not even error prints. That doesn't seem to be correct.

If you could point me to someone who maybe responsible, I will gladly ask my question there. It's very hard to understand the connections between esp-idf, arduino, pioarduino and platform io for someone not involved in the development and lacking some skills.

Jason2866, 2025-03-24:
> (Also just for my curiosity: why are you precompiling hte libs? Error tracking would be easier, when having the source code locally and in the compile chain.)

That's the way espressif32 Arduino works.

Jason2866, 2025-03-24:
You use the weird Arduino Nano Board. Only the makers - Arduino.cc can tell how to get serial output. But there you want get an answer either, since latest official Arduino core for this board is the old outdated core 2.0.17. It seems the makers of the board can't handle there weird Hardware/Software design, since the stay at core 2.0.x 
If you want to use actual Arduino core, don't waste your time and use any other esp32x board. Not the Arduino Nano.
Clearly speaking you walk alone with the Arduino Nano board and Core 3.x.

Jason2866, 2025-03-24:
> What I'm asking is: How to track those errors down? As I understand, the libraries are compiled, and there are no debug prints coming from that; not even error prints. That doesn't seem to be correct.

For debug output from source code you have to compile the precompiled static code blobs yourself. Not usual. Has to be done with https://github.com/espressif/esp32-arduino-lib-builder

Kybernaut000, 2025-03-24:
Thank you very much for your quick response. I'm sorry for bothering you with this stuff. I read about the root cause of pioarduino, it was some trouble with Espressif if I understood it right. Your project pioarduino is very important for a lot of people, and you are doing a great job.

About the Arduino Nano ESP32: It's quite disappointing that this board never got the attention it deserved from Arduino. It feels like they didn't think it to the end. And the Arduino IDE - that's a menace to society, not an IDE.

I'm using this board because I needed a little more RAM and flash for my current project, but it will be my first and last Arduino Nano ESP32. I've worked with some ESP32 DevKit, and had no (like in: none, absolutely zero) problems.

Jason2866, 2025-03-24:
RAM is always the same. There is no difference between the boards using the same MCU.
There are S3 devkit boards with 32 MB flash. In other words there are plenty of boards to choose from with flash sizes between 4 and 32MB.