pioarduino/platform-espressif32 #127: Internal compiler error when compiling a library with pioarduino/platform-espressif32

dejwk, 2025-03-24:
I am consistently getting an internal compiler error when building one of my libraries on pioarduino. The same problem never occurred with platformio. I can reproduce it 100% reliably; it fails for this library:

https://github.com/dejwk/roo_windows

I tried several different releases, to the same effect.

Have anyone seen this before? Is it possible that pioarduino uses an outdated version of gcc, maybe?

during RTL pass: ira
In file included from /home/dawidk/Documents/roo/roo_windows/src/roo_windows/core/container.h:11,
                 from /home/dawidk/Documents/roo/roo_windows/src/roo_windows/core/panel.h:9,
                 from /home/dawidk/Documents/roo/roo_windows/src/roo_windows/core/task.h:6,
                 from /home/dawidk/Documents/roo/roo_windows/src/roo_windows/core/activity.h:5,
                 from /home/dawidk/Documents/roo/roo_windows/src/roo_windows.h:2,
                 from src/main.cpp:17:
/home/dawidk/Documents/roo/roo_windows/src/roo_windows/core/widget.h: In member function 'virtual roo_windows::Padding roo_windows::Widget::getPadding() const':
/home/dawidk/Documents/roo/roo_windows/src/roo_windows/core/widget.h:395:59: internal compiler error: in elimination_costs_in_insn, at reload1.cc:3539
  395 |   virtual Padding getPadding() const { return Padding(0); }
      |                                                           ^
0x7f412c66cd67 __libc_start_call_main
        ../sysdeps/nptl/libc_start_call_main.h:58
0x7f412c66ce24 __libc_start_main_impl
        ../csu/libc-start.c:360
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.

Jason2866, 2025-03-24:
Arduino core 3.x has breaking changes. You have to adopt your code. Not a pioarduino issue.

closed 2025-03-24 by Jason2866

dejwk, 2025-03-24:
Please let me point to this part of the error message:

(...): internal compiler error: in elimination_costs_in_insn, at reload1.cc:3539

It points to an internal source file of the gcc compiler. So, yes, the root of the problem is a compiler bug - not this repository.

That said, I have found no recent hits for this bug on the Internet, but there are some old ones, e.g. https://gcc.gnu.org/bugzilla/show_bug.cgi?id=85034

Hence my suspicion that somehow this is related to an outdated version of the compiler.

For years, I have had no problems compiling this code with PlatformIO, using the default toolchains, as well as with Arduino IDE (including the recent version of the Arduino IDE, 2.3.4, which uses IDF 5.1 and Arduino 3.0). The only time it shows up, consistently, is with compilers/toolchains shipped with pioarduino.

Are you absolutely positive that this can't be some sort of configuration issue, with pioarduino picking up something in a version that is either outdated, or, to the contrary, very recent and not yet stable?

Jason2866, 2025-03-24:
pioarduino use exactly the same toolchains as ArduinoIDE for the core versions.

Jason2866, 2025-03-24:
Most probably it is this compiler bug https://github.com/jcmvbkbc/gcc-xtensa/issues/14#issuecomment-1995272776