pioarduino/platform-espressif32 #128: ESP-S3 - Debugger not starting (maybe was #36) - some progress

mhaberler, 2025-03-26:
Platform: MacOS M1, Vscode
Arduino 3.2 rc2

had the following situation several times, did not see a clear pattern why (it's not Arduino versions):

- cd to the project dir in terminal
- start VSCode with `code .`
- build/flash works perfectly fine
- starting debug session - Debug Console shows: 'Could not start debugger process > Error: spawn platformio ENOENT' 

It turns out that if you activate the pio python environment before starting Vscode with `code .`, then debugging works fine

NB this has no effect on other aspects of the build - all working fine despite NOT activating the pio Python environment

````
MacBookPro networking/cores3-read-qrcode logcanvas $ pio
zsh: command not found: pio
MacBookPro networking/cores3-read-qrcode logcanvas $ pio
zsh: command not found: pio
MacBookPro networking/cores3-read-qrcode logcanvas $ source /Users/mah/.platformio/penv/bin/activate
(penv) MacBookPro networking/cores3-read-qrcode logcanvas $ pio
Usage: pio [OPTIONS] COMMAND [ARGS]...

Options:
  --version          Show the version and exit.
...
  upgrade   Upgrade PlatformIO Core to the latest version
(penv) MacBookPro networking/cores3-read-qrcode logcanvas $ code .
````

now starting a debug session works fine, even with latest 3.2rc2

TLDR; activate the platformio Python environment before starting VSCode 

Michael

Jason2866, 2025-03-26:
Thx a lot for this finding. Will investigate how to implement the activating of the pio python environment without doing harm elsewhere ;-)

Jason2866, 2025-03-27:
Honestly currently i have no idea to fix this nice without (ugly) side effects. The Python process `spawn` needs env var set to work. Dont want to modify the system env, which is a can of worms to set for all platforms and unset when closing VSC.
Another workaround is to install Platformio in the default system Python. For calling the Debugger it is fine to have a version which is in path

mhaberler, 2025-03-27:
not a big issue as there is a workaround

I'm willing to try a few things like Python Environments on VSCode

how would I install pioarduino from the command line?

Jason2866, 2025-03-27:
We both use a Apple Silicon Mac so this should work
```
pip install -U https://github.com/Jason2866/platformio-core/archive/refs/tags/v6.1.18.zip --break-system-packages
```

mhaberler, 2025-03-27:
will report

mhaberler, 2025-03-28:
semi-related: it seems the MacOS Python version must be the 3.11 Framework version (whose install package is not online anymore on python.org)

I tried the 3.13 Framework and headed straight to the pits

Jason2866, 2025-03-28:
Using python v3.13 with Mac (and in CI runs) works without issues.

mhaberler, 2025-03-29:
do you mean v3.13 installed as default Python (Framework)? this is after which I got stuck

and afaict, the pioarduino plugin then installs its own version anyway?

Jason2866, 2025-03-29:
Can't reproduce. Works when system uses 3.13 and/ or Platformio python env is 3.13

mhaberler, 2025-03-29:
no, that was 3.11

I'll dump 3.11 and switch to 3.13m long overdue anyway
Time machine will save me just in case ;)

Jason2866, 2025-03-29:
It is a strange issue. This simple test sketch does not show the error. Debugger starts correctly.
https://github.com/Jason2866/test-pio-debug

mhaberler, 2025-04-02:
I tried your repo 

first, it initially did start the debugger

it did not stop at `setup` - however, with `tbreak app_main` it stops there (old but unrelated)
also old but unrelated: trying to terminate a stuck debug session may leave Python instances hanging in the background 
example after exiting VSCode (pretty sure this is a MacOS only issue):

`````
MacBookPro networking/test-pio-debug main $ ps -ax |grep Pyth
29704 ??         0:54.29 /Library/Frameworks/Python.framework/Versions/3.13/Resources/Python.app/Contents/MacOS/Python /Users/mah/.platformio/penv/bin/platformio debug -e esp32s3 --load-mode manual --interface gdb --interpreter=mi2 -q
`````

I also noticed if you change the VSCode profile the spawn error can be made to reappear

I had some success by wiping `.vscode` and starting over

the causal chain is still unclear to me

Jason2866, 2025-04-02:
I think the issue is caused by not activating the Python venv. It looks this works most of the time.
But this is not the way a Python venv should be used. My googling all ended that this error ("spawn") is typical for an not activated Python env. More precisely the not set PATH env variable. In my tests it was enough to set the PATH env to the venv Python to make the error go away. Using a Python venv seems to work most of the time... It is a Platformio concept design bug. Since it is horrible to implement to activate and deactivate a Python venv correctly for all OS i leave this faulty behaviour as it is. 

TLDR The Python venv needs to be activated to use the Debugger.

Jason2866, 2025-08-03:
@mhaberler You can try latest release https://github.com/pioarduino/platform-espressif32/releases/download/55.03.30-2/platform-espressif32.zip it does set the Python path during runtime it may solve the spawn error. The Python path settings done are close to activating a Python venv

mhaberler, 2025-08-03:
ok will do, thanks!

mhaberler, 2025-08-03:
puzzling:
cat idf-env.json
{
    "idfInstalled": {
        "/Users/mah/.platformio/packages/tool-esp_install-v5.4": {
            "version": "5.4",                <------------------ isnt that supposed to be 5.5?
            "path": "/Users/mah/.platformio/packages/tool-esp_install",
            "features": [
                "core"
            ],
            "targets": [
                "esp32c2",
                "esp32c5",
                "esp32s3",
                "esp32h2",
                "esp32c61",
                "esp32s2",
                "esp32c3",
                "esp32c6",
                "esp32h21",
                "esp32p4",
                "esp32h4"
            ]
        }
    }
}%

Jason2866, 2025-08-03:
it is correct, tool-esp_install version has nothing to do with IDF