pioarduino/platform-espressif32 #132: Release v3.2.0 based on ESP-IDF v5.4.1 - Platformio upload task, output handling

ikorman, 2025-03-31:
Sorry if this is wrong forum, but I just installed v3.2.0 based on ESP-IDF v5.4.1 and noticed change in Upload task

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1-n16r8v.html
PLATFORM: Espressif 32 (54.3.20) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.2.0
 - framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a
 - tool-esptoolpy @ 4.8.8
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 3.2.0
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-riscv32-esp @ 14.2.0+20241119
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP_SSLClient @ 2.1.16
|-- WiFi @ 3.2.0
Building in release mode
Retrieving maximum program size .pio\build\stable\firmware.elf
Checking size .pio\build\stable\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.0% (used 45772 bytes from 327680 bytes)
Flash: [=         ]  15.0% (used 981478 bytes from 6553600 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM8
Uploading .pio\build\stable\firmware.bin
esptool.py v4.8.8
Serial port COM8
Connecting....
```
Here it stops with messages, but upload is performed and when finished just push out rest of the messages all at once (part below). Before it was following progress of the update (that percentage part). If this is not related to this forum, pls be kind and let me know where to report this.

```
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: cc:8d:a2:ec:c3:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected flash size: 16MB
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000fffff...
SHA digest in image updated
Compressed 20208 bytes to 13058...
Writing at 0000000000... (100 %)
Wrote 20208 bytes (13058 compressed) at 0x00000000 in 0.5 seconds (effective 295.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0000008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 305.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 000000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 437.3 kbit/s)...
Hash of data verified.
Compressed 981872 bytes to 634907...
Writing at 0000010000... (2 %)
Writing at 000001c47f... (5 %)
Writing at 00000265b2... (7 %)
Writing at 000002e1e5... (10 %)
Writing at 000003c791... (12 %)
Writing at 000004475b... (15 %)
Writing at 0000049cb4... (17 %)
Writing at 000004f512... (20 %)
Writing at 0000054aa7... (23 %)
Writing at 000005a34d... (25 %)
Writing at 000005fb5c... (28 %)
Writing at 000006556e... (30 %)
Writing at 000006a6b1... (33 %)
Writing at 000006facf... (35 %)
Writing at 0000074ef2... (38 %)
Writing at 000007a026... (41 %)
Writing at 000007f854... (43 %)
Writing at 0000084afd... (46 %)
Writing at 0000089df7... (48 %)
Writing at 000008f275... (51 %)
Writing at 0000094815... (53 %)
Writing at 0000099de1... (56 %)
Writing at 000009f43e... (58 %)
Writing at 00000a464f... (61 %)
Writing at 00000a9526... (64 %)
Writing at 00000ae8ec... (66 %)
Writing at 00000b3a22... (69 %)
Writing at 00000b8d92... (71 %)
Writing at 00000bddc7... (74 %)
Writing at 00000c3008... (76 %)
Writing at 00000c8fc2... (79 %)
Writing at 00000ce6c4... (82 %)
Writing at 00000d3ee9... (84 %)
Writing at 00000dec24... (87 %)
Writing at 00000e5072... (89 %)
Writing at 00000ea041... (92 %)
Writing at 00000ef706... (94 %)
Writing at 00000f54fe... (97 %)
Writing at 00000fb041... (100 %)
Wrote 981872 bytes (634907 compressed) at 0x00010000 in 9.7 seconds (effective 807.7 kbit/s)...
Hash of data verified.
Hard resetting via RTS pin...
```

Jason2866, 2025-03-31:
It is the changed behaviour from the used latest esptool.py.

closed 2025-03-31 by Jason2866

ikorman, 2025-03-31:
I don't get it why, but OK ...

ikorman, 2025-04-01:
Well, as I did find this very annoying, did take a look at esptool.py command. Apparently this issue is in logger.py in method 

**def print_overwrite(self, message: str, last_line: bool = False):**

I have added **sys.stdout.flush()** at the end, and now it's working fine.
```

    def print_overwrite(self, message: str, last_line: bool = False):
        """
        Print a message, overwriting the currently printed line.

        If last_line is False, don't append a newline at the end
        (expecting another subsequent call will overwrite this one).

        After a sequence of calls with last_line=False, call once with last_line=True.

        If output is not a TTY (for example redirected a pipe),
        no overwriting happens and this function is the same as print().
        """
        if hasattr(sys.stdout, "isatty") and sys.stdout.isatty():
            # ansi_clear clears the line to prevent artifacts from previous lines
            print(
                f"\r{self.ansi_clear}{message}",
                end="\n" if last_line else "",
                flush=True,
            )
        else:
            print(message)
            sys.stdout.flush()
```

Again, if this is not right repo to report this, pls let me know where to do it.

Thx

Jason2866, 2025-04-01:
Thx, the esptool version is a fork too. Thx for the code to change. Will test and use when working. The fork for esptool is here https://github.com/pioarduino/esptool

Jason2866, 2025-04-01:
New pioarduino esptoolpy v4.8.9 released with your change. New version is used in all devel branches already
Can be forced to use in `platformio.ini` with this entry
```
platform_packages = tool-esptoolpy @ https://github.com/pioarduino/esptool/releases/download/v4.8.9/esptool.zip
```