pioarduino/platform-espressif32 #133: esp32-s3-devkitc1-n16r16.json - wrong board name

ikorman, 2025-04-01:
Just a small correction - for board esp32-s3-devkitc1-n16r16, there is wrong name defined in .json file, so it's presented in PIO board selection as Espressif ESP32-S3-DevKitC-1-N16R8V

```
{
  "build": {
    "arduino": {
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_ESP32S3_DEV",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "psram_type": "opi",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "default_tool": "esp-builtin",
    "onboard_tools": [
      "esp-builtin"
    ],
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "platforms": [
    "espressif32"
  ],
  "name": "Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB Flash Quad, 16 MB PSRAM Octal)",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html",
  "vendor": "Espressif"
}
```

Jason2866, 2025-04-02:
If you want to see it changed please provide a PR

ikorman, 2025-04-02:
Yeap, you are right, I was just wondering how the process looks like. 

Thx, will take a look. I thought because it's error, it's handled as issue.

ikorman, 2025-04-02:
done, hope it's OK

ikorman, 2025-04-02:
Oh man... :) I hope now is OK.

closed 2025-04-02 by Jason2866

FunDeckHermit, 2025-04-02:
Congratulations on your first contribution to Open Source!