pioarduino/platform-espressif32 #137: board upload offset address is ignored

m-mcgowan, 2025-04-16:
### PROBLEM DESCRIPTION

`board_upload.offset_address` is not being used during the upload.

### TO REPRODUCE

Using release 54.03.20. (also present in 51.03.04)

```
# platformio.ini
board_upload.offset_address = 0x800000
board_upload.flash_size = 16MB
board_build.partitions = partition.csv
```

Partition table derived from `default_8MB.csv` with an additional factory partition at 8MB.
```
# partition.csv
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x330000,
app1,     app,  ota_1,   0x340000,0x330000,
spiffs,   data, spiffs,  0x670000,0x180000,
coredump, data, coredump,0x7F0000,0x10000,
factory,  app,  factory, 0x800000,0x330000

```

Running `pio run -t upload` ends with

```
Wrote 861472 bytes (447141 compressed) at 0x00010000 in 6.1 seconds (effective 1128.2 kbit/s)...
```

The default application address (ota_0) is used rather than the specified offset.

### EXPECTED BEHAVIOUR

After performing an upload, the address written to is the one given in `platform.ini`.

### SCREENSHOTS

Nothing to see here, I guess I could have pasted in a sad kitten.

### ADDITIONAL CONTEXT

Looking through `builder/main.py` it does grab the offset address from `platform.ini`, but then scans the partitions, and overrides the address with the start address of  the `ota_0` partition.  I would like instead for it to write to the `factory` partition.

I'm grateful for the effort put into this project, it has helped me out a lot.  I'd be very happy to submit a PR once I'm clear on the existing requirements and/or get a green light for my suggested approach below:

I'm thinking of introducing a new variable

`board_upload.partition=<partition name>` 

and then using that to find the correct address from the partition table, if it is defined, otherwise defaulting to the `ota_0` partition to preserve the current behavior.

Alternatively, if `board_upload.offset_address` is defined, then use that unconditionally, without parsing the partition table.

All things being equal, I'd prefer the first option (partition name), so that the offset address isn't repeated between `platformio.ini` and the partition table, but that's a loosely held opinion, and secondary to any overriding concerns.

Guidance on the best way forward very much welcomed.

**(Please, remember to close the issue when the problem has been addressed)**

sivar2311, 2025-04-16:
Hi @m-mcgowan 

Please see https://community.platformio.org/t/esp32-the-program-size-is-greater-than-maximum-allowed/47068/2?u=sivar2311

m-mcgowan, 2025-04-16:
Thank you for the pointer.  As the thread mentions (and from what I've seen in the code), `build.app_partition_name` is presently used to determine the maximum upload size, but doesn't influence the upload partition.

This could be extended to also be the default target partition, or, better put, the default value for `build_upload.partition`. 

So, we'd have these rules:

1. `build.app_partition_name` continues to function as it does presently, helping compute the maximum app size.
1. `build_upload.partition_name` defines which partition to flash to, defaulting to `build.app_partition_name`, or `ota_0` when both `build.app_partition_name` and `build_upload.partition_name` are not not defined. (Preserving existing behavior to flash to `ota_0` by default.)
1. `board_upload.offset_address` takes absolute precedence, and when defined sets the target address. The script could print a warning if the offset address doesn't correspond to the start of a partition, or when `build_upload.partition_name` and `board_upload.offset_address` are both defined.   This is a departure from the current behavior, where the script would always flash to `ota_0` even with an address offset given - that change in behavior would address the bug mentioned here.

How does that sound?

Jason2866, 2025-04-16:
Not a bug, it does not work as you think.

Jason2866, 2025-04-16:
Btw a factory image at 0x800000 is not a good idea. esp32x tries to start, when a lot has been broken, as last try a firmware at 0x10000. So a recovery firmware should always be placed at 0x10000

m-mcgowan, 2025-04-16:
I tried with the latest release, and as far as I can tell, the upload will not write to a different partition - it always writes to `ota_0`.  The code in `main.py` is specifically looking for `ota_0` irrespective of the settings for offset address or `build.app_partition_name` (which, as you know, isn't used during upload.)  This shortcoming is what I hope to address here, so that the target upload partition can be specified in a way that's backwardly compatible with how things currently work.

> Define a boards json to add the extra image. Name the partitions accordingly in partitions.csv so later on it can be used to identify.

I haven't needed to define a new board.json file so far, which seems like a much heavier lift than just setting `board_build.partitions` in `platformio.ini` to point to the partition file. Is defining a new board.json is necessary? If so, please explain more.

Ultimately the devices will have several pre-installed apps (e.g. recovery app, production run tests and diagnostics.)  It would be simplest to use PlatformIO to build and flash these various apps to their corresponding partitions, rather than diving down into the IDF tools.